project-chip/connectedhomeip #11720: followup: we should reserve some read handlers for reads.

yunhanw-google, 2021-11-12:
This makes sense for a subscribe, but should a regular read just return Busy?  Or is it that too many subscriptions can keep reads from working?  If so, followup: we should reserve some read handlers for reads.

_Originally posted by @bzbarsky-apple in https://github.com/project-chip/connectedhomeip/pull/11667#discussion_r747747362_

bzbarsky-apple, 2021-11-12:
@mrjerryjohns @tcarmelveilleux This seems pretty critical to do: persistent subscriptions should not be able to completely lock out reads, which right now they can.

@franck-apple Do we have a label or project of some sort tracking "must happen for 1.0 to have sanity" work that is not really part of a bigger project?  I guess I'll toss in "spec" for now....

mrjerryjohns, 2021-11-12:
Yes, agreed. I'd recommend we reserve no more than 2 handlers (preferably, 1), since reads are not long-lived exchanges, and the chances of coincidental reads from multiple peers simultaneously is low.

Cases where that might not be true based on past experience would be a highly correlated event that affects multiple clients in the system simultaneously (like resumption from a comms outage, or power outage). I'm not certain we should over-design for that scenario anyways, since recovery from those usually takes time, and clients would have to be built to be defensive towards those cases anyways.

bzbarsky-apple, 2021-11-12:
Yes, the reservation should be pretty small.  Read attempts that run into no handlers get a BUSY status and can try again...

PR #12764 (2021-12-09): Reserve the last read handler for ReadInteraction
#### Problem
Fix #11720 

#### Change overview
Reserve the last read handler for ReadInteraction

#### Testing
Update the unit test that return error when subscribe tries to grab the last readHandler
closed 2021-12-09 by yunhanw-google