project-chip/connectedhomeip #11757: EADDRNOTAVAIL on mdns socket.

cecille, 2021-11-12:
#### Problem
[1636749473.538079][803313:803318] CHIP:DIS: Attempt to mDNS broadcast failed:  ../../src/inet/UDPEndPoint.cpp:1042: OS Error 0x02000063: Cannot assign requested address

Andrei suggests that the link is not yet up when we try to send and waiting after the setup results in a useable socket. 

 #### Proposed Solution
There must be a way to tell when a socket is usable, right? We should do that.

andy31415, 2021-11-12:
using `ip a` shows tentative at start and becomes global after a sleep  of 2 seconds.

Unsure how to figure out if an IP is valid or not and if we can do this at runtime at all. It sounds like bind succeeds (so we use inetaddr_any) however sending fails because local ip is not yet fully known/fixed.

cecille, 2022-09-19:
Going to leave this open until 1.0 settles and we can come back and see if this is till an issue. I suspect this may come back to bite us later, so let's leave it open for now so it doesn't get lost.

closed 2022-09-27 by stale[bot]