project-chip/connectedhomeip #11760: [ota-provider-app] Support for error/timeout midway in transfer and later resume

ukatira, 2021-11-12:
#### Problem
There is no way right now to trigger an error/timeout midway in OTA file transfer.

 #### Proposed Solution
Add support for error/timeout midway in OTA file transfer in the ota-provider-app to verify requestor handles these correctly and can resume transfer successfully.

Section 11.20.3.5 Pg 665: Idle time-out SHALL be no less than 5 minutes for either Receiver-Driver or Asynchronous mode, before aborting a transfer.
Section 11.20.3.5 Pg 666: The ReceiveInit message from the OTA Requestor MAY have the RC[STARTOFS] bit set and associated STARTOFS field set to indicate the resumption of a transfer previously aborted, or to affect partial windowed access to the portion of a Software Image desired.

yufengwangca, 2022-02-02:
@holbrookt could you help to take a look?

holbrookt, 2022-02-02:
This seems necessary for cert and 1.0. Can you confirm @ukatira ?

ukatira, 2022-02-02:
I think Resume functionality got moved to post V1.0 so this will be needed when Requestors can support resuming a transfer previously aborted.

carol-apple, 2022-02-17:
I think even without resume supported, this part can still be validated:

> Section 11.20.3.5 Pg 665: Idle time-out SHALL be no less than 5 minutes for either Receiver-Driver or Asynchronous mode, before aborting a transfer.

closed 2022-03-05 by woody-apple