project-chip/connectedhomeip #11762: TC-DIAGETH-1.2 Ethernet Diagnostics Cluster Commands Always Return 0

cm-s-dev, 2021-11-12:
#### Problem
`packet-tx-count` and `packet-rx-count` always return 0. Other attributes may also be affected.
Maybe there is some issue when multiple network interfaces are available?

#### Commands
`chip-tool  ethernetnetworkdiagnostics read  packet-rx-count 1234 0`
`chip-tool  ethernetnetworkdiagnostics read  packet-tx-count 1234 0`

 #### Test Setup:
Tried testing with RPI+Ubuntu Desktop together and separately, but still same results.
Previously this worked with te6.

`chip-tool` and `chip-all-clusters-app`: 
commit `f51eeb39de1e58c8f55a13cfc59782e8e11c6948 (HEAD -> test_event_7, origin/test_event_7)`

#### RPI Logs (both apps):
[TC-DIAGETH-1.2_fail.txt](https://github.com/project-chip/connectedhomeip/files/7530975/TC-DIAGETH-1.2_fail.txt)

holbrookt, 2022-02-01:
@cm-s-dev is this still an issue? Please close if not.

cm-s-dev, 2022-03-09:
Above commands seem to be working fine in `5cbe2931f64e16a44fc843e148325909fbe1511d (tag: TE8/rc3)`. Closing.

closed 2022-03-09 by cm-s-dev