project-chip/connectedhomeip #11774: K32W061(k32w) lighting failed to receive read command from chip-tool

TE-N-ElvenWang, 2021-11-15:
Based on **_test_event_6_** branch, on the Linux OTBR device, provision the k32w lighting successfully. 
"$chip-tool onoff toggle 1" will be able to toggle the LED on/off.

But when try to read the values from k32w lighting device, it failed.

> [220342][P][ZCL]OpCreds: Writing fabric into attribute store at index 0: fabricId 0x0000000000000000, nodeId 0xDA52C299A445651E vendorId 0x000E
[220355][P][ZCL]OpCreds: Stored hu fabrics in fabrics list attribute.
[220362][D][DMG]AttributePath is not interested
[220367][P][ZCL]Unknown cluster ID: 0x0000_003E
[220371][D][DMG]AttributePath is not interested
[220376][P][ZCL]Unknown cluster ID: 0x0000_003E
[223501][P][IN]Build plaintext message 0x400cdc8 to 0x0000000000000000 of type 49 and protocolId 0 on exchange 15071i with MessageCounter 2986198637.
[223515][P][IN]Sending plaintext msg 0x400cdc8 to 0x0000000000000000 at utc time: ld msec
[223523][P][IN]Sending msg on generic transport
[223529][D][DL]Thread packet SENT: UDP, len 758
....
....
....
[230433][D][EM]Handling via exchange: 15072r, Delegate: 0x0x400a190
[230440][D][-]Receive Read request
[230443][D][DMG]IM RH moving to [Initialized]
[230448][D][DMG]IM RH moving to [GeneratingReports]
[230453][D][IN]Received a duplicate message with MessageCounter: 1707369858
[230460][P][EM]Received message of type 0x32 with vendorId 0x0000 and protocolId 0x0000 on exchange 15071r
[230470][D][EM]Handling via exchange: 15071r, Delegate: 0x0
[230476][E][EM]**OnMessageReceived failed, err = Error CHIP:0x0000002A**
[230483][D][IN]Received a duplicate message with MessageCounter: 0
[230489][P][EM]Received message of type 0x02 with vendorId 0x0000 and protocolId 0x0001 on exchange 15072r
[230500][D][IN]Secure message was encrypted: Msg ID 1
[230505][P][IN]Build encrypted message 0x4025c48 to 0x000000000001B669 of type 16 and protocolId 0 on exchange 15072i with MessageCounter 1.
[230518][P][IN]Sending encrypted msg 0x4025c48 to 0x000000000001B669 at utc time: ld msec
[230526][P][IN]Sending msg on generic transport
[230531][D][DL]Thread packet SENT: UDP, len 82

Attached the logs for chip-tool and k32w:
[linux_chip-tool.log](https://github.com/project-chip/connectedhomeip/files/7537125/linux_chip-tool.log)
[k32w_lighting.log](https://github.com/project-chip/connectedhomeip/files/7537126/k32w_lighting.log)

cecille, 2022-02-01:
Is this still an issue on more recent code?

woody-apple, 2022-02-08:
This is indeed old enough, we should re-verify this - as I believe lighting is generally working now. Please re open if needed! @TE-N-ElvenWang

closed 2022-02-08 by woody-apple