project-chip/connectedhomeip #11821: [TC-WIFIDIAG-1.1] WifiNetworkDiagnostics cluster attributes report incorrect values

andrew-lifx, 2021-11-16:
#### Problem
Information reported from some of the WifiNetworkDiagnostics cluster attributes is incorrect.
* Bssid is empty
* Security type is correct
* WifiVersion potentially wrong expected 2 Reported 0.
* Wifi Channel is correct
* RSSI looks OK.
* Beacon Count is 0 (?)
* All packet counts and rates are 0. Expect some non-zero values.

Looks like some attributes may be simply returning the default value.

#### Test Cases
TC-WIFIDIAG-1.1 | Attributes with server as DUT
* https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/wifi_diagnostics.adoc#tc-wifidiag-1-1-attributes-with-server-as-dut

#### Logs
[21-11-16_04:07_TC-WIFIDIAG-1.1.log](https://github.com/project-chip/connectedhomeip/files/7543591/21-11-16_04.07_TC-WIFIDIAG-1.1.log)

#### DUT
 * ESP32 Platform
 * On/Off Light Switch

franck-apple, 2022-01-22:
@andrew-lifx and @wqx6 , is this fixed by PR #12082?

wqx6, 2022-01-24:
@franck-apple Yes, it is fixed now.

closed 2022-01-24 by chshu