project-chip/connectedhomeip #11822: General Diagnostics Cluster networlk-interfaces attribute returns no data

andrew-lifx, 2021-11-16:
#### Problem
Test Event 7, TC-IDM-2.2 failed.
Reading the networlk-interfaces attribute of the General Diagnostics Cluster return an empty data element.

An array of device interfaces was expected as per the example in the test Step for the Test Case.
#### Test Cases
TC-IDM-2.2 | Report Data Action from DUT to TH

#### Logs
[221-11-16_01:49_TC-IDM-2.2.log](https://github.com/project-chip/connectedhomeip/files/7543788/221-11-16_01.49_TC-IDM-2.2.log)

#### DUT
* commit f51eeb39de1e58c8f55a13cfc59782e8e11c6948 (HEAD -> test_event_7, origin/test_event_7
 * ESP32 Platform
 * On/Off Light Switch

fessehaeve, 2021-11-16:
Try a different attribute which has values. For eg `./chip-tool operationalcredentials read fabrics-list 12344321 0`

andrew-lifx, 2021-11-17:
Yes that works. And if if had used that the test case would have passed.
But the issue is not about the test case failing. It is about the fact that there is no data for the network-interfaces.
It most like a platform integration issue, as it apparently works on the RPi.

franck-apple, 2022-01-22:
@andrew-lifx and @wqx6 , is this fixed by PR #12082?

wqx6, 2022-01-24:
@franck-apple Yes, it was fixed now.

closed 2022-01-24 by chshu