project-chip/connectedhomeip #11827: #TE7 [TC-TSUIC-2.1] Wrong verification steps

fessehaeve, 2021-11-16:
#### Problem
The verification steps for [TC-TSUIC-2.1] are wrong. They seem to be copied from the Thermostat cluster test when in fact this test is for Thermostat User Configuration cluster.

 #### Proposed Solution
It should be something like this:

   ```
 "# TC-TSUIC-2.1 2"
    "./chip-tool thermostatuserinterfaceconfiguration read temperature-display-mode 1 1"
    "./chip-tool thermostatuserinterfaceconfiguration read keypad-lockout 1 1"

    "# TC-TSUIC-2.1 3a"
    "./chip-tool thermostatuserinterfaceconfiguration write temperature-display-mode 1 1 1"
    "./chip-tool thermostatuserinterfaceconfiguration write keypad-lockout 1 1 1"
    
    "# TC-TSUIC-2.1 3b"
    "./chip-tool thermostatuserinterfaceconfiguration read temperature-display-mode 1 1"
    "./chip-tool thermostatuserinterfaceconfiguration read keypad-lockout 1 1"

    "# TC-TSUIC-2.1 4"
    "./chip-tool thermostatuserinterfaceconfiguration read schedule-programming-visibility 1 1"

    "# TC-TSUIC-2.1 5a"
    "./chip-tool thermostatuserinterfaceconfiguration write schedule-programming-visibility 1 1 1"
    
    "# TC-TSUIC-2.1 5b"
    "./chip-tool thermostatuserinterfaceconfiguration read schedule-programming-visibility 1 1"
```

woody-apple, 2022-02-08:
@krypton36 do you know where this should be filed?

fessehaeve, 2022-02-09:
This is in the test verification Excel sheet. The error is still there on TE7.5 test verification document.

krypton36, 2022-03-30:
> @krypton36 do you know where this should be filed?

@woody-apple I apologize for the late response, but this issue should be file under [chip-test-pans](https://github.com/CHIP-Specifications/chip-test-plans).

closed 2022-06-07 by woody-apple