project-chip/connectedhomeip #11828: #TE7 [TC-GENDIAG-1.1] and [TC-GENDIAG-1.2] description and verification steps are wrong.

fessehaeve, 2021-11-16:
#### Problem

The description for these tests is very misleading and incorrect. The test as the name says is general and doesn’t depend on the networking capability of the DUT. In addition, the verification steps are wrong.
```
3.1.1. [TC-GENDIAG-1.1] Attributes with server as DUT
	Purpose						
	This test case verifies the behavior of the non-global attributes of the WiFi network diagnostics cluster server.	
```					
and
```
	3.1.2. [TC-GENDIAG-1.2] Attributes with client as DUT					Pls refer to the PICS definition https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/general_diagnostics.adoc	
	Purpose						
	This test case verifies the behavior of the non-global attributes of the ethernet network diagnostics cluster server.	
```					

 #### Proposed Solution

Description should be updated from the PICS doc.
Verification steps for [TC-GENDIAG-1.1] should be something like this:

```
  "# TC-GENDIAG-1.1"
"./chip-tool generaldiagnostics read network-interfaces 1 0"
"./chip-tool generaldiagnostics read reboot-count 1 0"
```

mrjerryjohns, 2022-01-31:
Is this still relevant?

mrjerryjohns, 2022-01-31:
@fessehaeve can you confirm?

fessehaeve, 2022-02-01:
@mrjerryjohns the verification steps are correct now. The description is still wrong.
TC-GENDIAG-1.1
`This test case verifies the behavior of the non-global attributes of the WiFi network diagnostics cluster server.`
TC-GENDIAG-1.2
`This test case verifies the behavior of the non-global attributes of the ethernet network diagnostics cluster server.`

woody-apple, 2022-02-08:
@raju-apple Do you know where this should go?

closed 2022-06-14 by woody-apple