project-chip/connectedhomeip #11850: Update ClusterBase Associate method for group

jepenven-silabs, 2021-11-16:
#### Problem
Current implementation is tailor made for testing group communication.

 #### Proposed Solution
make this function more versatile by handling all possible scenario. For example : 
- Handle the fact that we may not have a CASE session to the device.

turon, 2022-02-01:
Reference to code in question: https://github.com/project-chip/connectedhomeip/blob/ddb7ce76a2459b7d3a2f9434dbf86c7e5b1a6a6d/examples/chip-tool/templates/partials/test_cluster.zapt#L270

It looks like both group and non-group cases are handled, so perhaps this can be closed:

```
 {{#if isGroupCommand}}
        cluster.AssociateWithGroup({{>device}}, groupId);
 {{else}}
        cluster.Associate({{>device}}, endpoint);
{{/if}}
```

Removing tag `v1_triage_split_4`

jepenven-silabs, 2022-02-01:
@turon Wrong code quoted.

Problem is still present in master. The code assumes that the `DeviceProxy` object will have a valid SecureSession which is always true in the context of a `yaml` test but won't be in other context (e.g. inside Chip-tool ? ).

So this issue is still valid and still need to be fixed

```
CHIP_ERROR ClusterBase::AssociateWithGroup(DeviceProxy * device, GroupId groupId)
{
    // TODO Update this function to work in all possible conditions Issue #11850

    CHIP_ERROR err = CHIP_NO_ERROR;

    mDevice = device;
    if (mDevice->GetSecureSession().HasValue())
    {
        // Local copy to preserve original SessionHandle for future Unicast communication.
        Optional<SessionHandle> session = mDevice->GetExchangeManager()->GetSessionManager()->CreateGroupSession(
            groupId, mDevice->GetSecureSession().Value()->AsSecureSession()->GetFabricIndex());
        // Sanity check
        if (!session.HasValue() || !session.Value()->IsGroupSession())
        {
            err = CHIP_ERROR_INCORRECT_STATE;
        }

        mGroupSession.Grab(session.Value());
    }
    else
    {
        // something fishy is going on
        err = CHIP_ERROR_INCORRECT_STATE;
    }

    // Set to 0 for now.
    mEndpoint = 0;

    return err;
}
```

jepenven-silabs, 2022-02-01:
This might need to be fixed before #1818

jepenven-silabs, 2022-02-28:
Not needed after all. #1818 was done without this changes.

PR #16583 (2022-03-25): [Group] Remove AssociateWithGroup
#### Problem
Fix #11850

#### Change overview
Remove the `AssociateWithGroup` method since it didn't make sense to tie a `DeviceProxy` object with a group command. The api can now support a WriteAttribute command without any `DeviceProxy` object involved.

However, for the yaml test part of it, the necessary data is still retrieved from the DeviceProxy for convenience reasons only since this is limited to test code.

#### Testing
Yaml test.
closed 2022-03-25 by bzbarsky-apple