project-chip/connectedhomeip #11857: chip-control app heap usage increased drastically in TE7 branch

hhuang1911, 2021-11-16:
#### Problem
We had an issue on the TE7 chip-tool build, the heap size increased from 2MB to 151MB comparing to TE6 version. 
TE7 build on RPI4:
==5551== HEAP SUMMARY:
==5551== in use at exit: 121,468 bytes in 1,312 blocks
==5551== total heap usage: 18,031 allocs, 16,719 frees, 151,176,146 bytes allocated
TE6 build on RPI4:
==4382== HEAP SUMMARY:
==4382== in use at exit: 114,475 bytes in 1,241 blocks
==4382== total heap usage: 12,362 allocs, 11,121 frees, 2,023,257 bytes allocated

*Expected behavior: 
We should be conscious about memory usage even for a controller implementation. The heap usage should be tracked between releases. 
*Actual behavior:
Due to this memory jump, we cannot run it on our controller implementation due to limited RAM
*Steps to reproduce
Run with valgrind for pair a IPv4 device using chip-tool. 
[valgrind-TE6.txt](https://github.com/project-chip/connectedhomeip/files/7548547/valgrind-TE6.txt)
[valgrind-TE7.txt](https://github.com/project-chip/connectedhomeip/files/7548549/valgrind-TE7.txt)

* System configuration: RPi4 with TE release images

 #### Proposed Solution
Identify the cause of heap usage jump and reduce it
Include a validation step to check the heap usage in each release

andy31415, 2022-02-03:
This may also be a result of moving some pools onto heap instead of static memory. I am not convinced this is something we should fix even for 1.x based on the given report.

hhuang1911, 2022-02-04:
A big memory jump like this cannot be ignored or pushed away without detailed analysis report. There are controllers running on devices other than phone or pad. 150MB memory is not something like a TV or a STB can easily give away.

closed 2022-10-09 by stale[bot]