project-chip/connectedhomeip #11895: [TC-I-2.2] Identify-time attribute is not counting down

andrew-lifx, 2021-11-17:
#### Problem
The identify-time attribute is not counting down while an identity is active. It remains at the last set value either by and identify command or an attribute write.

#### Test Cases
TC-I-2.2 | Primary functionality with server as DUT

#### Logs
[21-11-16_23:19:49_TC-I-2.2.log](https://github.com/project-chip/connectedhomeip/files/7551078/21-11-16_23.19.49_TC-I-2.2.log)

#### DUT
* commit f51eeb39de1e58c8f55a13cfc59782e8e11c6948 (HEAD -> test_event_7, origin/test_event_7
 * ESP32 Platform
 * On/Off Light Switch

franck-apple, 2022-01-22:
@andrew-lifx and @wqx6 , is this fixed by PR #12082?

wqx6, 2022-01-24:
@franck-apple Yes, it is fixed now.

closed 2022-01-24 by chshu