project-chip/connectedhomeip #11896: [TC-I-2.2] DUT is send default Failure response to identify-query when identify is not active

andrew-lifx, 2021-11-17:
#### Problem
From Test Event 7
The DUT is generating a default Failure response to an identify-query when identify is not active on the device.
Not response is expected from a device that is not actively identifying.

#### Test Cases
TC-I-2.2 | Primary functionality with server as DUT

#### Logs
[21-11-16_23:19:49_TC-I-2.2.log](https://github.com/project-chip/connectedhomeip/files/7551078/21-11-16_23.19.49_TC-I-2.2.log)

#### DUT
* commit f51eeb39de1e58c8f55a13cfc59782e8e11c6948 (HEAD -> test_event_7, origin/test_event_7
 * ESP32 Platform
 * On/Off Light Switch

 #### Proposed Solution
According to the Specification the "If the device is not currently identifying itself it SHALL take no further action".

fessehaeve, 2021-11-17:
Agree. The verification step should look like this:

```
cmd: ./chip-tool identify identify-query 12344321 1
[1637082466.508251][31167:31167] CHIP:CTL: Generating NOC
[1637082466.508734][31167:31167] CHIP:CTL: Generating ICAC
[1637082466.514293][31167:31167] CHIP:DL: AUDIT: ===== RANDOM NUMBER GENERATOR AUDIT START ====
[1637082466.514338][31167:31167] CHIP:DL: AUDIT: * Validate buf1 and buf2 are <<<different every run/boot!>>>
[1637082466.514351][31167:31167] CHIP:DL: AUDIT: * Validate r1 and r2 are <<<different every run/boot!>>>
[1637082466.514365][31167:31167] CHIP:DL: AUDIT: * buf1: EC4D03FC2998C4CC037C74C5CE99B2B3
[1637082466.514377][31167:31167] CHIP:DL: AUDIT: * buf2: 23D141AD308F4C92B0F7FEE396465C43
[1637082466.514433][31167:31167] CHIP:DL: AUDIT: * r1: 0x58EF47B2 r2: 0xEBFFE27A
[1637082466.514444][31167:31167] CHIP:DL: AUDIT: ===== RANDOM NUMBER GENERATOR AUDIT END ====
[1637082466.515014][31167:31167] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-KJKjPI)
[1637082466.515623][31167:31167] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1637082466.515670][31167:31167] CHIP:DL: NVS set: chip-counters/reboot-count = 86 (0x56)
[1637082466.516596][31167:31167] CHIP:DL: Got Ethernet interface: eth0
[1637082466.517152][31167:31167] CHIP:DL: Found the primary Ethernet interface:eth0
[1637082466.517769][31167:31167] CHIP:DL: Got WiFi interface: wlan0
[1637082466.517844][31167:31167] CHIP:DL: Failed to reset WiFi statistic counts
[1637082466.517892][31167:31167] CHIP:IN: UDP::Init bind&listen port=5541
[1637082466.518000][31167:31167] CHIP:IN: UDP::Init bound to port=5541
[1637082466.518022][31167:31167] CHIP:IN: UDP::Init bind&listen port=5541
[1637082466.518105][31167:31167] CHIP:IN: UDP::Init bound to port=5541
[1637082466.518125][31167:31167] CHIP:IN: TransportMgr initialized
[1637082466.518187][31167:31167] CHIP:DIS: Init fabric pairing table with server storage
[1637082466.518536][31167:31167] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-SnKTI3)
[1637082466.528827][31167:31167] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1637082466.528893][31167:31167] CHIP:DL: NVS set: chip-counters/GlobalMCTR = 86000 (0x14FF0)
[1637082466.529041][31167:31167] CHIP:CTL: System State Initialized...
[1637082466.536438][31167:31167] CHIP:DL: Avahi client registered
[1637082466.538973][31167:31167] CHIP:ZCL: Using ZAP configuration...
[1637082466.540246][31167:31167] CHIP:DIS: Verifying the received credentials
[1637082466.542816][31167:31167] CHIP:DIS: Added new fabric at index: 1, Initialized: 1
[1637082466.542862][31167:31167] CHIP:DIS: Assigned compressed fabric ID: 0x6AE7E3C75FBC2D4D, node ID: 0x000000000001B669
[1637082466.542882][31167:31167] CHIP:CTL: Joined the fabric at index 1. Compressed fabric ID is: 0x6AE7E3C75FBC2D4D
[1637082466.543364][31167:31172] CHIP:DL: CHIP task running
[1637082466.543601][31167:31172] CHIP:DL: HandlePlatformSpecificBLEEvent 32785
[1637082466.543635][31167:31172] CHIP:TOO: Sending command to node 0xbc5c01
[1637082466.544502][31167:31172] CHIP:IN: Prepared plaintext message 0xaaaaeae57810 to 0x0000000000000000 of type 0x30 and protocolId (0, 0) on exchange 50323i with MessageCounter:3430839734.
[1637082466.544566][31167:31172] CHIP:IN: Sending plaintext msg 0xaaaaeae57810 with MessageCounter:3430839734 to 0x0000000000000000 at monotonic time: 354131419 msec
[1637082466.544763][31167:31172] CHIP:SC: Sent Sigma1 msg
[1637082466.917077][31167:31172] CHIP:EM: Received message of type 0x31 with protocolId (0, 0) and MessageCounter:4167468942 on exchange 50323i
[1637082466.917525][31167:31172] CHIP:EM: Rxd Ack; Removing MessageCounter:3430839734 from Retrans Table on exchange 50323i
[1637082466.917775][31167:31172] CHIP:EM: Removed CHIP MessageCounter:3430839734 from RetransTable on exchange 50323i
[1637082466.918174][31167:31172] CHIP:SC: Received Sigma2 msg
[1637082466.918577][31167:31172] CHIP:SC: Peer assigned session session ID 86
[1637082466.925439][31167:31172] CHIP:SC: Sending Sigma3
[1637082466.926546][31167:31172] CHIP:EM: Piggybacking Ack for MessageCounter:4167468942 on exchange: 50323i
[1637082466.926636][31167:31172] CHIP:IN: Prepared plaintext message 0xaaaaeae57810 to 0x0000000000000000 of type 0x32 and protocolId (0, 0) on exchange 50323i with MessageCounter:3430839735.
[1637082466.926697][31167:31172] CHIP:IN: Sending plaintext msg 0xaaaaeae57810 with MessageCounter:3430839735 to 0x0000000000000000 at monotonic time: 354131801 msec
[1637082466.926947][31167:31172] CHIP:SC: Sent Sigma3 msg
[1637082467.408208][31167:31172] CHIP:EM: Received message of type 0x40 with protocolId (0, 0) and MessageCounter:4167468943 on exchange 50323i
[1637082467.408350][31167:31172] CHIP:EM: Rxd Ack; Removing MessageCounter:3430839735 from Retrans Table on exchange 50323i
[1637082467.408392][31167:31172] CHIP:EM: Removed CHIP MessageCounter:3430839735 from RetransTable on exchange 50323i
[1637082467.409280][31167:31172] CHIP:SC: Success status report received. Session was established
[1637082467.409533][31167:31172] CHIP:IN: New secure session created for device 0x0000000000BC5C01, key 86!!
[1637082467.409715][31167:31172] CHIP:TOO: Sending cluster (0x00000003) command (0x00000001) on endpoint 1
[1637082467.410057][31167:31172] CHIP:DMG: ICR moving to [AddingComm]
[1637082467.410268][31167:31172] CHIP:DMG: ICR moving to [AddedComma]
[... cut ...]
[1637082467.778513][31167:31167] CHIP:CTL: Shutting down the System State, this will teardown the CHIP Stack
[1637082467.778904][31167:31167] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-0wq0Mf)
[1637082467.779816][31167:31167] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1637082467.779877][31167:31167] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1637082467.779894][31167:31167] CHIP:DL: Inet Layer shutdown
[1637082467.779983][31167:31167] CHIP:DL: BLE layer shutdown
[1637082467.780004][31167:31167] CHIP:DL: System Layer shutdown
```

franck-apple, 2022-01-22:
@andrew-lifx and @wqx6 , is this fixed by PR #12082?

gjc13, 2022-01-24:
Fix verified.

closed 2022-01-24 by gjc13

wqx6, 2022-01-24:
Yes, it was fixed now. @franck-apple