project-chip/connectedhomeip #11899: {TC-GENDIAG-1.1] All generaldiagnostics cluster attributes returned zero or empty data

andrew-lifx, 2021-11-17:
#### Problem
From Test Event 7
The DUT returned zero or an empty list/string/array for all cluster attributes in [TC-GENDIAG-1.1]
This implies that none of these attributes are actually being filled out with valid data.

#### Test Cases
TC-GENDIAG-1.1 |  Attributes with server as DUT

#### Logs
[21-11-17_04:06:53_TC-GENDIAG-1.1.log](https://github.com/project-chip/connectedhomeip/files/7551626/21-11-17_04.06.53_TC-GENDIAG-1.1.log)

#### DUT
* commit f51eeb39de1e58c8f55a13cfc59782e8e11c6948 (HEAD -> test_event_7, origin/test_event_7
 * ESP32 Platform
 * On/Off Light Switch, all-cluster-app

wqx6, 2021-11-18:
Hi, Andrew, Here is a temporary fix for your issues https://github.com/wqx6/connectedhomeip/tree/test_event_7 , I will raise a PR later.

bzbarsky-apple, 2022-05-05:
@andrew-lifx Is this still an issue?

franck-apple, 2022-05-17:
resolved in #11654

closed 2022-05-17 by franck-apple