project-chip/connectedhomeip #11909: Add Device Attestation validation to `chip-cert` tool for YAML test support

tcarmelveilleux, 2021-11-17:
#### Problem
There is no "freestanding" implementation of the Device Attestation validation procedure that could be executed by implementers and by CSG test harness based on responses obtained, to ascertain the current validity of a given implementation.

 #### Proposed Solution
In order to automate Device Attestation Procedure testing, we could have a command-line tool that can validate the AttestationResponse and and provide a result and logs, so that it can be wrapped in a delegated YAML command.

This will allow executing an instrumented view of the device attestation procedure. 

The tooling could then be replaced in the future with an equivalent independent implementation.

Two commands are necessary:

- Validation of AttestationSignature on AttestationResponse
- Validation of AttestationSignature on NOCSRResponse

The common arguments for both:

- PAI cert in DER, encoded as hex
- DAC cert in DER, encoded as hex
- Attestation Challenge from secure session, encoded as hex
- Nonce (AttestationNonce, CSRNonce), encoded as hex
- AttestationSignature (from either AttestationResponse or NOCSRResponse), encoded as hex

The specific arguments are either:

- AttestationElements from AttestationResponse
- NOCSRElements from NOCSRResponse

holbrookt, 2022-01-31:
@tcarmelveilleux This seems important - should it remain in the 1.0 list?

naveenkommareddi, 2022-11-02:
Impact: 8 Device Attestation TCs
Priority: High

tcarmelveilleux, 2023-11-27:
This is done.

closed 2023-11-27 by tcarmelveilleux