project-chip/connectedhomeip #11929: #TE7 TC-TSTAT-2.1 and TC-TSTAT-2.2 fail because of a hardcoded feature map

fessehaeve, 2021-11-17:
#### Problem
TC-TSTAT-2.1 and TC-TSTAT-2.2 fail because of a hardcoded feature map and related pre attribute changed callback `MatterThermostatClusterServerPreAttributeChangedCallback`. With the current implementation if a device doesn't have the `OccupiedCoolingSetpoint` attribute all write commands fail.

 #### Proposed Solution
Make implementation independent of feature map.

chrisbeach59, 2021-11-22:
Feature map is only hard coded due to the incorrect implementation in generated code by ZAP of the feature map.  Once feature map is working correctly, the over ride in the code is simply commented out, and the feature map is then used as intended.

fessehaeve, 2021-11-23:
Putting the feature map problem aside the fix also breaks things if one doesn't play with the thermostat cluster implementation. The feature map problem is in many other clusters too and solving it this way doesn't improve the state of things. I made my above mentioned tests fail so that we can track this issue properly.

chrisbeach59, 2021-11-23:
Attached are 2 log files for 2.1 and 2.2.  All cooling attributes and auto related attributes have been removed. 
The Feature map was set to HEAT only.  
Test scripts still attempt to acccess/mutate the cooling related attributes, which fail as expected.  
The results are as expected.  

I have verified with others on the SDK team that Feature Map is being used with it's intended purpose.

[heat-only-2-1-log.txt](https://github.com/project-chip/connectedhomeip/files/7588282/heat-only-2-1-log.txt)
[heat-only-2-2-log.txt](https://github.com/project-chip/connectedhomeip/files/7588284/heat-only-2-2-log.txt)
[thermostat.zap.txt](https://github.com/project-chip/connectedhomeip/files/7588349/thermostat.zap.txt)

fessehaeve, 2021-11-23:
I am just saying the implementation should have ignored checking feature map until it was supported properly.

chrisbeach59, 2021-11-23:
TOT seems Feature Map seems to be working correctly, now.  
I will re-test with the #define overide disabled in the thermostat-server.cpp file and if all OK will do a PR
Sorry for any confusion.

closed 2021-12-15 by chrisbeach59