project-chip/connectedhomeip #11945: Power attribute missing in Pump Configuration and Control Cluster in Mac Chip Tool

Garik098, 2021-11-17:
The below mentioned attributes were present in ZCL but for MAC Chip Tool they are not available to read.

zclread PumpConfigurationAndControl MinConstPressure 12344321 1 0
zclread PumpConfigurationAndControl MaxConstPressure 12344321 1 0
zclread PumpConfigurationAndControl MinCompPressure 12344321 1 0
zclread PumpConfigurationAndControl MaxCompPressure 12344321 1 0
zclread PumpConfigurationAndControl MinConstSpeed 12344321 1 0
zclread PumpConfigurationAndControl MaxConstSpeed 12344321 1 0
zclread PumpConfigurationAndControl MinConstFlow 12344321 1 0
zclread PumpConfigurationAndControl MaxConstFlow 12344321 1 0
zclread PumpConfigurationAndControl MinConstTemp 12344321 1 0
zclread PumpConfigurationAndControl MaxConstTemp 12344321 1 0
zclread PumpConfigurationAndControl PumpStatus 12344321 1 0
zclread PumpConfigurationAndControl Speed 12344321 1 0
zclread PumpConfigurationAndControl LifetimeRunningHours 12344321 1 0
zclread PumpConfigurationAndControl Power 12344321 1 0
zclread PumpConfigurationAndControl LifetimeEnergyConsumed 12344321 1 0
zclread PumpConfigurationAndControl ControlMode 12344321 1 0

Attaching a screenshot of what attributes were currently available for the Mac Chip Tool to read. Other than these attributes others were not present to read.
[image]

Garik098, 2022-01-25:
Lifetimerunninghours and Power are now appearing in the chip tool list

turon, 2022-02-01:
Closing based on @Garik098 's comment.
Can reopen if in fact the support has not been added fully.

closed 2022-02-01 by turon