project-chip/connectedhomeip #11960: need extend supersetof function to overlapped for clusterInfoA and clusterInfoB, and need to get the overlapped part, and send report with overlapped part, perhaps in

yunhanw-google, 2021-11-18:
yep, need extend supersetof function to overlapped for clusterInfoA and clusterInfoB, and need to get the overlapped part, and send report with overlapped part, perhaps in

_Originally posted by @yunhanw-google in https://github.com/project-chip/connectedhomeip/pull/11795#discussion_r751958515_

tcarmelveilleux, 2022-01-31:
@erjiaqing Is this complete?

erjiaqing, 2022-02-08:
No, this is not complete, but I think this is a optimization for saving CPU cycles and memory consumption.

My idea would be adding a filter to `AttributePathExpandIterator` instead of calculating the overlapped paths.

woody-apple, 2022-02-08:
Let's move this to post v1.X until it's clear this is required.

yunhanw-google, 2022-05-23:
https://github.com/project-chip/connectedhomeip/pull/18597 where we resolve partial overlap issue between subscribed path and setDirty path

closed 2022-05-23 by yunhanw-google