project-chip/connectedhomeip #11999: [AMBD] generaldiagnostics network-interfaces

chrisbeach59, 2021-11-18:
#### Problem
Empty list is returned
chip-tool generaldiagnostics read network-interfaces 1111 0
ReportData =
[1637275283.644095][30454:30459] CHIP:DMG: {
[1637275283.644159][30454:30459] CHIP:DMG: 	AttributeDataList =
[1637275283.644223][30454:30459] CHIP:DMG: 	[
[1637275283.644284][30454:30459] CHIP:DMG: 		AttributeDataElement =
[1637275283.644345][30454:30459] CHIP:DMG: 		{
[1637275283.644408][30454:30459] CHIP:DMG: 			AttributePath =
[1637275283.644487][30454:30459] CHIP:DMG: 			{
[1637275283.644589][30454:30459] CHIP:DMG: 				NodeId = 0x457,
[1637275283.644816][30454:30459] CHIP:DMG: 				EndpointId = 0x0,
[1637275283.644891][30454:30459] CHIP:DMG: 				ClusterId = 0x33,
[1637275283.644975][30454:30459] CHIP:DMG: 				FieldTag = 0x0000_0000,
[1637275283.645072][30454:30459] CHIP:DMG: 			}
[1637275283.645161][30454:30459] CHIP:DMG: 				
[1637275283.645241][30454:30459] CHIP:DMG: 			Data = [
[1637275283.645308][30454:30459] CHIP:DMG: 				
[1637275283.645426][30454:30459] CHIP:DMG: 			],
[1637275283.645498][30454:30459] CHIP:DMG: 			DataElementVersion = 0x0,
[1637275283.645574][30454:30459] CHIP:DMG: 		},
[1637275283.645650][30454:30459] CHIP:DMG: 		
[1637275283.645723][30454:30459] CHIP:DMG: 	],
[1637275283.645796][30454:30459] CHIP:DMG: 	
[1637275283.645853][30454:30459] CHIP:DMG: }
[1637275283.646014][30454:30459] CHIP:ZCL: ReadAttributesResponse:
[1637275283.646074][30454:30459] CHIP:ZCL:   ClusterId: 0x0000_0033
[1637275283.646122][30454:30459] CHIP:ZCL:   attributeId: 0x0000_0000
[1637275283.646164][30454:30459] CHIP:ZCL:   status: Success                (0x0000)
[1637275283.646207][30454:30459] CHIP:ZCL:   attribute TLV Type: 0x16
[1637275283.646271][30454:30459] CHIP:TOO: OnGeneralDiagnosticsNetworkInterfacesListAttributeResponse: 0 entries

 #### Proposed Solution
<suggested fix, suggested enhancement>

pankore, 2021-11-26:
Fixed by https://github.com/project-chip/connectedhomeip/pull/12076

yufengwangca, 2022-02-11:
Fixed by https://github.com/project-chip/connectedhomeip/pull/12076

closed 2022-02-11 by yufengwangca