project-chip/connectedhomeip #12011: [TE#7][TC-LVL-3.1] move operation LevelControl exceeding the max level

Kxuan, 2021-11-19:
#### Problem
The step 2 in the test plan says:
```
2: TH reads CurrentLevel attribute from DUT
Verify that the DUT response indiicates that the CurrentLevel attribute has the value recorded in 1a, or if the MaxLevel attribute is not supported (PICS_MAXLEVEL = false) 0xFE
```
But SDK acturally raise the current level to 0xff, which is too large.

logs from DUT; 
```
I (185761) chip[EM]: Received message of type 0x10 with protocolId (0, 0) and MessageCounter:1 on exchange 42936r
I (185771) chip[ZCL]: Event: move from 254
I (185781) chip[ZCL]:  to 255
I (185781) chip[ZCL]: (diff +1)
I (185781) app-devicecallbacks: PostAttributeChangeCallback - Cluster ID: '0x0008', EndPoint ID: '0x01', Attribute ID: '0x0000'
```
 #### Proposed Solution
limit the max level to 254

jmartinez-silabs, 2022-01-22:
Let me confirm on monday, but this is fixed.

jmartinez-silabs, 2022-01-27:
Confirmed move command works correctly and stops at the 254 limit not 255

fixed with #13338

closed 2022-01-27 by jmartinez-silabs