project-chip/connectedhomeip #12020: [TE7][TC-MF-1.19] Wrong Status Code when opening commissioning window when commissionee already has max number of fabrics

du48s03, 2021-11-19:
#### Problem
 * expected behavior: If a commissionee already has the max number of fabrics, if an admin tries to open a commission window for it, the response code should be TableFull (0x05)
 * actual behavior: The status code was Success (0x00), although any following pairing attempt will fail.

turon, 2022-02-01:
Commissioning should return helpful error codes in v1.
Removing tag `v1_triage_split_4`.

bzbarsky-apple, 2022-05-05:
@du48s03 The spec absolutely allows opening a commissioning window when the max number of fabrics is already present.   The new commissioner can then choose whether it should remove an existing fabric before adding its fabric or do something else.

The only thing in the spec that returns TableFull is AddNOC.

So the behavior here is exactly following the spec.

closed 2022-05-05 by bzbarsky-apple