project-chip/connectedhomeip #12026: [TC-OO-2.3]off-with-effect and  on-with-timed-off  are not supported with 0x81 status

Minan-Tuyasmart, 2021-11-19:
Problem：
[TC-OO-2.3]off-with-effect and  on-with-timed-off  are not supported with 0x81 status

Steps to reproduce the problem:
platform:

OTBR base on matter_te7 branch (72bb3d45684f837e67e7f1ec20a9a8e3ac4a4419).

lighting-app run on the EFR32MG24 board base on commit 82bcd4e

Testing image on RasPi：raspi_test_event_7_shrunk.zip

run command as follow  
'''
ubuntu@matter-te-7:~/connectedhomeip$ ./out/standalone/chip-tool onoff off-with-effect 0 0 12344321 1
[1637310381.688554][3949:3949] CHIP:CTL: Generating NOC
[1637310381.688920][3949:3949] CHIP:CTL: Generating ICAC
[1637310381.694083][3949:3949] CHIP:DL: AUDIT: ===== RANDOM NUMBER GENERATOR AUDIT START ====
[1637310381.694130][3949:3949] CHIP:DL: AUDIT: * Validate buf1 and buf2 are <<<different every run/boot!>>>
[1637310381.694156][3949:3949] CHIP:DL: AUDIT: * Validate r1 and r2 are <<<different every run/boot!>>>
[1637310381.694181][3949:3949] CHIP:DL: AUDIT: * buf1: 2269A7E857C5F2E6222B9B7390F10D31
[1637310381.694203][3949:3949] CHIP:DL: AUDIT: * buf2: 57E60FF7A9B786EAB63D5C5FC13616A8
[1637310381.694269][3949:3949] CHIP:DL: AUDIT: * r1: 0x02DE5D2E r2: 0x1A40A880
[1637310381.694293][3949:3949] CHIP:DL: AUDIT: ===== RANDOM NUMBER GENERATOR AUDIT END ====
[1637310381.694858][3949:3949] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-WCTqHv)
[1637310381.695497][3949:3949] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1637310381.695555][3949:3949] CHIP:DL: NVS set: chip-counters/reboot-count = 6 (0x6)
[1637310381.696406][3949:3949] CHIP:DL: Got Ethernet interface: eth0
[1637310381.696965][3949:3949] CHIP:DL: Found the primary Ethernet interface:eth0
[1637310381.697524][3949:3949] CHIP:DL: Got WiFi interface: wlan0
[1637310381.699589][3949:3949] CHIP:DL: Found the primary WiFi interface:wlan0
[1637310381.699654][3949:3949] CHIP:IN: UDP::Init bind&listen port=5541
[1637310381.699773][3949:3949] CHIP:IN: UDP::Init bound to port=5541
[1637310381.699801][3949:3949] CHIP:IN: UDP::Init bind&listen port=5541
[1637310381.699893][3949:3949] CHIP:IN: UDP::Init bound to port=5541
[1637310381.699917][3949:3949] CHIP:IN: TransportMgr initialized
[1637310381.699983][3949:3949] CHIP:DIS: Init fabric pairing table with server storage
[1637310381.700342][3949:3949] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-nHK8WE)
[1637310381.700895][3949:3949] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1637310381.700944][3949:3949] CHIP:DL: NVS set: chip-counters/GlobalMCTR = 6000 (0x1770)
[1637310381.701068][3949:3949] CHIP:CTL: System State Initialized...
[1637310381.709678][3949:3949] CHIP:DL: MDNS failed to join multicast group on veth6af6f6b for address type IPv4: ../../examples/chip-tool/third_party/connectedhomeip/src/inet/IPEndPointBasis.cpp:387: Inet Error 0x00000110: Address not found
[1637310381.712191][3949:3949] CHIP:DL: MDNS failed to join multicast group on vethfe40d16 for address type IPv4: ../../examples/chip-tool/third_party/connectedhomeip/src/inet/IPEndPointBasis.cpp:387: Inet Error 0x00000110: Address not found
[1637310381.714119][3949:3949] CHIP:DL: MDNS failed to join multicast group on wpan0 for address type IPv4: ../../examples/chip-tool/third_party/connectedhomeip/src/inet/IPEndPointBasis.cpp:387: Inet Error 0x00000110: Address not found
[1637310381.714292][3949:3949] CHIP:ZCL: Using ZAP configuration...
[1637310381.714921][3949:3949] CHIP:DIS: Verifying the received credentials
[1637310381.716982][3949:3949] CHIP:DIS: Added new fabric at index: 1, Initialized: 1
[1637310381.717026][3949:3949] CHIP:DIS: Assigned compressed fabric ID: 0x547D9076FA9D4399, node ID: 0x000000000001B669
[1637310381.717051][3949:3949] CHIP:CTL: Joined the fabric at index 1. Compressed fabric ID is: 0x547D9076FA9D4399
[1637310381.717419][3949:3954] CHIP:DL: CHIP task running
[1637310381.717526][3949:3954] CHIP:TOO: Sending command to node 0xbc5c01
[1637310381.718272][3949:3954] CHIP:IN: Prepared plaintext message 0xaaaafdbe6860 to 0x0000000000000000 of type 0x30 and protocolId (0, 0) on exchange 52191i with MessageCounter:1462489186.
[1637310381.718332][3949:3954] CHIP:IN: Sending plaintext msg 0xaaaafdbe6860 with MessageCounter:1462489186 to 0x0000000000000000 at monotonic time: 2655871 msec
[1637310381.718517][3949:3954] CHIP:SC: Sent Sigma1 msg
[1637310383.870003][3949:3954] CHIP:EM: Received message of type 0x31 with protocolId (0, 0) and MessageCounter:1954715492 on exchange 52191i
[1637310383.870136][3949:3954] CHIP:EM: Rxd Ack; Removing MessageCounter:1462489186 from Retrans Table on exchange 52191i
[1637310383.870338][3949:3954] CHIP:EM: Removed CHIP MessageCounter:1462489186 from RetransTable on exchange 52191i
[1637310383.870410][3949:3954] CHIP:SC: Received Sigma2 msg
[1637310383.870469][3949:3954] CHIP:SC: Peer assigned session session ID 7
[1637310383.878356][3949:3954] CHIP:SC: Sending Sigma3
[1637310383.878929][3949:3954] CHIP:EM: Piggybacking Ack for MessageCounter:1954715492 on exchange: 52191i
[1637310383.878996][3949:3954] CHIP:IN: Prepared plaintext message 0xaaaafdbe6860 to 0x0000000000000000 of type 0x32 and protocolId (0, 0) on exchange 52191i with MessageCounter:1462489187.
[1637310383.879052][3949:3954] CHIP:IN: Sending plaintext msg 0xaaaafdbe6860 with MessageCounter:1462489187 to 0x0000000000000000 at monotonic time: 2658031 msec
[1637310383.879275][3949:3954] CHIP:SC: Sent Sigma3 msg
[1637310384.644498][3949:3954] CHIP:EM: Received message of type 0x40 with protocolId (0, 0) and MessageCounter:1954715493 on exchange 52191i
[1637310384.644640][3949:3954] CHIP:EM: Rxd Ack; Removing MessageCounter:1462489187 from Retrans Table on exchange 52191i
[1637310384.644707][3949:3954] CHIP:EM: Removed CHIP MessageCounter:1462489187 from RetransTable on exchange 52191i
[1637310384.644780][3949:3954] CHIP:SC: Success status report received. Session was established
[1637310384.644829][3949:3954] CHIP:IN: New secure session created for device 0x0000000000BC5C01, key 7!!
[1637310384.644997][3949:3954] CHIP:TOO: Sending cluster (0x00000006) command (0x00000040) on endpoint 1
[... 138 lines cut ...]
[1637310426.578746][3957:3962] CHIP:DMG:
[1637310426.578799][3957:3962] CHIP:DMG:                },
[1637310426.578863][3957:3962] CHIP:DMG:
[1637310426.578906][3957:3962] CHIP:DMG:        ],
[1637310426.578951][3957:3962] CHIP:DMG:
[1637310426.578986][3957:3962] CHIP:DMG: }
[1637310426.579057][3957:3962] CHIP:DMG: Received Command Response Status for Endpoint=1 Cluster=0x0000_0006 Command=0x0000_0042 Status=0x81
[1637310426.579117][3957:3962] CHIP:TOO: Default Failure Response: 0x81
[1637310426.579208][3957:3962] CHIP:DMG: ICR moving to [AwaitingDe]
[1637310426.579307][3957:3962] CHIP:EM: Sending Standalone Ack for MessageCounter:1 on exchange 53453i
[1637310426.579434][3957:3962] CHIP:IN: Prepared encrypted message 0xffff837ed930 to 0x0000000000BC5C01 of type 0x10 and protocolId (0, 0) on exchange 53453i with MessageCounter:1.
[1637310426.579507][3957:3962] CHIP:IN: Sending encrypted msg 0xffff837ed930 with MessageCounter:1 to 0x0000000000BC5C01 at monotonic time: 2700732 msec
[1637310426.579713][3957:3962] CHIP:EM: Flushed pending ack for MessageCounter:1 on exchange 53453i
[1637310426.580063][3957:3957] CHIP:-: ../../examples/chip-tool/third_party/connectedhomeip/zzz_generated/chip-tool/zap-generated/cluster/Commands.h:45: CHIP Error 0x000000AC: Internal error at ../../examples/chip-tool/commands/common/CHIPCommand.cpp:84
[1637310426.580156][3957:3957] CHIP:TOO: Run command failure: ../../examples/chip-tool/third_party/connectedhomeip/zzz_generated/chip-tool/zap-generated/cluster/Commands.h:45: CHIP Error 0x000000AC: Internal error
[1637310426.703412][3957:3957] CHIP:CTL: Shutting down the System State, this will teardown the CHIP Stack
[1637310426.703770][3957:3957] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-T0jHSF)
[1637310426.704563][3957:3957] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1637310426.704629][3957:3957] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1637310426.704657][3957:3957] CHIP:DL: Inet Layer shutdown
[1637310426.705212][3957:3957] CHIP:DL: BLE layer shutdown
[1637310426.705241][3957:3957] CHIP:DL: System Layer shutdown
ubuntu@matter-te-7:~/connectedhomeip$

'''

jmartinez-silabs, 2021-12-07:
This should be fixed with 

https://github.com/project-chip/connectedhomeip/pull/11877

it is not in te7 branch

mkardous-silabs, 2021-12-09:
@jmartinez-silabs @Minan-Tuyasmart 
PR was merged. This issue can be closed.

closed 2021-12-14 by jmartinez-silabs