project-chip/connectedhomeip #12046: Manual Pairing Code "vid/pid present" bit not set for User-Intent Commissioning

CodeChronos928, 2021-11-19:
From `5.1.3.2. Custom Flow for Manual Pairing Code`:

> For User-intent commissioning flow and Custom Commissioning flow, the variant of Manual Pairing Code with Vendor ID and Product ID SHALL be used

Here, the code is only setting `vidPidPresentFlag` to true for `kCustom`, and not `kUserActionRequired`.

I wanted confirmation that this is a bug, and not a mis-reading of the spec, since it's somewhat confusing.  The spec calls the whole field `Custom Flow`.  I think that's a mistake too -- it should be `Commissioning Flow` (which is what it's called in the code).

https://github.com/project-chip/connectedhomeip/blob/8dc009bc31a377a60cffc3a457f885f82a5be7db/src/setup_payload/ManualSetupPayloadGenerator.cpp#L49

yufengwangca, 2022-02-02:
@cecille Could you help to take a look?

bzbarsky-apple, 2022-05-05:
Fixed in #15506.

closed 2022-05-05 by bzbarsky-apple