project-chip/connectedhomeip #12049: Assess the correct use of `ConcreteReadAttributePath` in the read call-path for the IM

mrjerryjohns, 2021-11-19:
This [PR](https://github.com/project-chip/connectedhomeip/pull/11705/files#r753406382) introduces a `ConcreteReadAttributePath` that contains a list index. However, it adds the use of that fairly late in the call chain for handling a read (inside the `ReadSingleClusterData` function). We should assess where the right place for this to be is.

mrjerryjohns, 2021-11-19:
Assigning this to you @erjiaqing since you're gonna be in this space a fair bit.

bzbarsky-apple, 2021-11-19:
Fwiw, I think `ReadSingleClusterData` should definitely take a `ConcreteReadAttributePath`.

bzbarsky-apple, 2021-11-20:
https://github.com/project-chip/connectedhomeip/pull/12067 fixes the obvious thing.  Need to figure out how to share the path-expanding iterator between read and write paths before we can push things further.

holbrookt, 2022-01-31:
@bzbarsky-apple @mrjerryjohns is this needed for 1.0? If not, please remove the label. Thanks!

bzbarsky-apple, 2022-02-01:
I don't think the remaining work here is a 1.0 blocker, personally.

mrjerryjohns, 2022-08-15:
Already done.

closed 2022-08-15 by mrjerryjohns