project-chip/connectedhomeip #12058: [TC-WIFIDIAG-3.1] reset-counts  command not supported in wifi diagnostics Cluster in TE7

praveenCY, 2021-11-20:
#### Problem
reset-counts  command not supported in wifi diagnostics Cluster

IMClusterCommandHandler.cpp doesnot have support for WiFiNetworkDiagnostics reset-counts

namespace WiFiNetworkDiagnostics {

void DispatchServerCommand(CommandHandler * apCommandObj, const ConcreteCommandPath & aCommandPath, TLV::TLVReader & aDataTlv)
{
    // We are using TLVUnpackError and TLVError here since both of them can be CHIP_END_OF_TLV
    // When TLVError is CHIP_END_OF_TLV, it means we have iterated all of the items, which is not a real error.
    // Any error value TLVUnpackError means we have received an illegal value.
    // The following variables are used for all commands to save code size.
    CHIP_ERROR TLVError = CHIP_NO_ERROR;
    bool wasHandled     = false;
    {
        switch (aCommandPath.mCommandId)
        {
        case Commands::ResetCounts::Id: {
            Commands::ResetCounts::DecodableType commandData;
            TLVError = DataModel::Decode(aDataTlv, commandData);
            if (TLVError == CHIP_NO_ERROR)
            {
                wasHandled = emberAfWiFiNetworkDiagnosticsClusterResetCountsCallback(apCommandObj, aCommandPath, commandData);
            }
            break;
        }
        default: {
            // Unrecognized command ID, error status will apply.
            ReportCommandUnsupported(apCommandObj, aCommandPath);
            return;
        }
        }
    }

Logs: [LOCK-TC-WIFIDIAG-3.1.txt](https://github.com/project-chip/connectedhomeip/files/7573473/LOCK-TC-WIFIDIAG-3.1.txt)

bzbarsky-apple, 2022-05-05:
@praveenCY Is this still an issue?

franck-apple, 2022-05-17:
dupe of #18321. Closing

closed 2022-05-17 by franck-apple