project-chip/connectedhomeip #12085: No response from DUT while running command [TC-LVL-2.2]

kvsmohan, 2021-11-22:
#### Problem
<No response from DUT while running command 
 While running the following command there is no response from DUT side but  the expected Outcome is Verify that the DUT response contains an integer. Record the value for usage in step 1b.

chip-tool levelcontrol read on-off-transition-time 1 0
chip-tool levelcontrol write on-off-transition-time 5 1 1
chip-tool levelcontrol read on-level 1 1
chip-tool levelcontrol read default-move-rate 1 1
chip-tool levelcontrol write default-move-rate 5 1 1
chip-tool levelcontrol write start-up-current-level 5  1 1
chip-tool levelcontrol read start-up-current-level 1 1

[[TC-LVL-2.2].txt](https://github.com/project-chip/connectedhomeip/files/7579131/TC-LVL-2.2.txt)

In DUT side there is no response

turon, 2022-02-01:
All cert tests must pass for v1.
Removing tag `v1_triage_split_4`

bzbarsky-apple, 2022-05-05:
@kvsmohan Is this an issue on tip?

naveenkommareddi, 2022-05-17:
Issue resolved per latest test reports

franck-apple, 2022-05-17:
Can be closed. Issue is resolved

closed 2022-05-17 by franck-apple