project-chip/connectedhomeip #12089: [TC-TSTAT-1.1] Value mismatch for FeatureMap attribute

manjunathambiger23, 2021-11-22:
Summary :
As per specification " default value of FeatureMap attribute is 0 but when we made a run the expected value is 11 " however in the test plan it reads the default value 0 as success response

Spec Screenshot: 
[image]

Testplan Screenshot:
[image]

Execution log :
[TSTAT.txt](https://github.com/project-chip/connectedhomeip/files/7579887/TSTAT.txt)

chrisbeach59, 2021-11-22:
Feature Map for thermostat is mandatory.  This is an issue with the spec.  I will put up a PR to fix this.  
The thermostat server uses it to know what the thermostat can and can not do.
At this time there is an issue with the generation of Feature map, for little endian machines.  As such there are overrides allowed in thermostat-server.cpp file to set the feature map of the thermostat.

chrisbeach59, 2021-11-22:
4.3.3.1. Features
This cluster SHALL support the FeatureMap bitmap attribute as defined below.

closed 2021-12-15 by chrisbeach59