project-chip/connectedhomeip #12090: [Multi Admin] TH2  commission fail

Suguotao, 2021-11-22:
#### Problem
dut:k32w061

<what's wrong or missing, please include any applicable:
 * expected behavior
  TH2 established successfully

 * actual behavior
  TH2 established failly 

 * steps to reproduce
 1. TH1 commissioned to dut
 2.TH1 commissioned successfully and send cmd(zcl AdministratorCommissioning OpenBasicCommissioningWindow 12344321 0 0 commissioningTimeout=360)
 3. dut return success 
 4. TH2 send cmd(connect -ip fdc2:589:2fa0:9a63:8592:7a2b:52a0:6b30 20202021 12344321) to commission to dut but fail

 * screenshots, images, logs, etc.
 
[DUT_k32w0.txt](https://github.com/project-chip/connectedhomeip/files/7578915/DUT_k32w0.txt)
[TH1_log.txt](https://github.com/project-chip/connectedhomeip/files/7578916/TH1_log.txt)
[TH2_log.txt](https://github.com/project-chip/connectedhomeip/files/7578917/TH2_log.txt)

fuxiaoming-lumi, 2021-11-26:
Let me add more test details.
Test Environment: **Matter light demo Thread Device(K32W06 DK)**, **Thread Border Router**, **2 admin Controller A&B (Raspbeerypi 4B+ with TE7)**
Firmware: Branch test_event_7 and patch e38b265

Issue:  
1. Pairing Device with Controller B(TH2) always fails in first time, but will succeed the second time if try again.
2. I can often see this error as following on the chip-device-ctrl side
> [1637902882.092410][18636:18641] CHIP:EM: OnMessageReceived failed, err = ../../src/messaging/ExchangeMgr.cpp:298: CHIP Error 0x0000002A: Invalid message type
[1637902882.129530][18636:18641] CHIP:EM: Received message of type 0x10 with protocolId (0, 0) and MessageCounter:1070751282 on exchange 57952i
[1637902882.129641][18636:18641] CHIP:EM: OnMessageReceived failed, err = ../../src/messaging/ExchangeMgr.cpp:270: CHIP Error 0x00000070: Unsolicited msg with originator bit clear
[1637902882.132342][18636:18641] CHIP:EM: Received message of type 0x10 with protocolId (0, 0) and MessageCounter:1070751283 on exchange 57952i
[1637902882.132430][18636:18641] CHIP:EM: OnMessageReceived failed, err = ../../src/messaging/ExchangeMgr.cpp:270: CHIP Error 0x00000070: Unsolicited msg with originator bit clear

Test Steps with python chip-device-ctrl:
1. **Controller A(TH1):** connect -ble 3840 20202021 12344321
2. **Controller A(TH1):** zcl NetworkCommissioning AddThreadNetwork 12344321 0 0 operationalDataset=hex:0e080000000000010000000300000c35060004001fffe0020811111111222222220708fd89cb00346132ee051000112233445566778899aabbccddeeff030e4f70656e54687265616444656d6f010212340410730771d0684557909e54a3554a1e4f5b0c0402a0fff8 breadcrumb=0 timeoutMs=1000
3. **Controller A(TH1):** zcl NetworkCommissioning EnableNetwork 12344321 0 0 networkID=hex:1111111122222222 breadcrumb=0 timeoutMs=1000
4. **Controller A(TH1):** close-ble
5. **Controller A(TH1):** resolve 12344321
6. **Controller A(TH1):** zcl AdministratorCommissioning OpenBasicCommissioningWindow 12344321 0 0 commissioningTimeout=360
7. **Controller B(TH2):**connect -ip fdc2:589:2fa0:9a63:8592:7a2b:52a0:6b30 20202021 12344321
8.  **Second Commissioning Failure** with Controller B(TH2)
9. Try the same command again.
10. **Controller B(TH2):** connect -ip fdc2:589:2fa0:9a63:8592:7a2b:52a0:6b30 20202021 12344321
11. **It will succeed this time.**

Log:

[TH1_log_20211126.txt](https://github.com/project-chip/connectedhomeip/files/7606424/TH1_log_20211126.txt)
[TH2_log_20211126.txt](https://github.com/project-chip/connectedhomeip/files/7606425/TH2_log_20211126.txt)
[DUT_k32w061.txt](https://github.com/project-chip/connectedhomeip/files/7606431/DUT_k32w061.txt)

tcarmelveilleux, 2022-01-31:
Failure of a test case on one platform. @doru91 please triage for K32W.

fuxiaoming-lumi, 2022-02-08:
Update the latest info. With TE7.5, this issue has not been reproduced but other issues were found.
Test Environment: Matter light demo Thread Device(K32W06 DK), Thread Border Router, 2 admin Controller A&B (Raspbeerypi 4B+ with TE7.5)
Firmware SDK: TE7.5 Branch b0af6ba42b6ce494748f18ecf3e054a400b8f45b

K32W06 platform does not support hardware acceleration engine. Currently, the whole commissioning time is up to 30 seconds and the establishment time of CASE secure channel is up to 10 seconds (with mbedtls library).

To solve this issue, the temporary solution needs to increase the timeout of chip-tool.
https://github.com/project-chip/connectedhomeip/blob/master/examples/chip-tool/commands/pairing/OpenCommissioningWindowCommand.h#L40

https://github.com/project-chip/connectedhomeip/blob/master/examples/chip-tool/commands/clusters/ModelCommand.h#L45

NXP team is still optimizing this issue.

franck-apple, 2022-05-18:
duplicate with #18084. Closing.

closed 2022-05-18 by franck-apple