project-chip/connectedhomeip #12124: [TC-MOD-2.1] ChangeToMode Verification Step 3 expects 0x85 as error response code but is getting 0xc6

Garik098, 2021-11-22:
In the Mode Select cluster test case TC-MOD-2.1 ChangeToMode Verification, in the last step (Step 3), the expectations is an error response with code 0x85. Even though I am seeing the error response, the error code is showing up to be 0xc6 and not the one expected.

Below is the log,

"[1637613691414] [25246:2371575] CHIP: [CTL] Generating NOC
[1637613691415] [25246:2371575] CHIP: [CTL] Generating ICAC
[1637613691415] [25246:2371575] CHIP: [DL] AUDIT: ===== RANDOM NUMBER GENERATOR AUDIT START ====
[1637613691415] [25246:2371575] CHIP: [DL] AUDIT: * Validate buf1 and buf2 are <<<different every run/boot!>>>
[1637613691415] [25246:2371575] CHIP: [DL] AUDIT: * Validate r1 and r2 are <<<different every run/boot!>>>
[1637613691415] [25246:2371575] CHIP: [DL] AUDIT: * buf1: 414B7AE819FDD429C3AA7ADBDA0AD446
[1637613691415] [25246:2371575] CHIP: [DL] AUDIT: * buf2: 065EA3162930BD63935F5929A39F597F
[1637613691415] [25246:2371575] CHIP: [DL] AUDIT: * r1: 0x30A12716 r2: 0x0382BBC1
[1637613691415] [25246:2371575] CHIP: [DL] AUDIT: ===== RANDOM NUMBER GENERATOR AUDIT END ====
[1637613691415] [25246:2371575] CHIP: [DL] _Init
[1637613691415] [25246:2371575] CHIP: [IN] UDP::Init bind&listen port=5541
[1637613691415] [25246:2371575] CHIP: [IN] UDP::Init bound to port=5541
[1637613691415] [25246:2371575] CHIP: [IN] UDP::Init bind&listen port=5541
[1637613691415] [25246:2371575] CHIP: [IN] UDP::Init bound to port=5541
[1637613691415] [25246:2371575] CHIP: [IN] TransportMgr initialized
[1637613691416] [25246:2371575] CHIP: [DIS] Init fabric pairing table with server storage
[1637613691416] [25246:2371575] CHIP: [CTL] System State Initialized...
[1637613691416] [25246:2371575] CHIP: [ZCL] Using ZAP configuration...
[1637613691416] [25246:2371575] CHIP: [DIS] Verifying the received credentials
[1637613691417] [25246:2371575] CHIP: [DIS] Added new fabric at index: 1, Initialized: 1
[1637613691417] [25246:2371575] CHIP: [DIS] Assigned compressed fabric ID: 0x56791D9DD63C9A95, node ID: 0x000000000001B669
[1637613691417] [25246:2371575] CHIP: [CTL] Joined the fabric at index 1. Compressed fabric ID is: 0x56791D9DD63C9A95
[1637613691417] [25246:2371579] CHIP: [DL] _OnPlatformEvent
[1637613691417] [25246:2371579] CHIP: [TOO] Sending command to node 0x1
[1637613691418] [25246:2371579] CHIP: [IN] Prepared plaintext message 0x7f8dce013280 to 0x0000000000000000 of type 0x30 and protocolId (0, 0) on exchange 42877i with MessageCounter:2136539305.
[1637613691418] [25246:2371579] CHIP: [IN] Sending plaintext msg 0x7f8dce013280 with MessageCounter:2136539305 to 0x0000000000000000 at monotonic time: 428590555 msec
[1637613691418] [25246:2371579] CHIP: [SC] Sent Sigma1 msg
[1637613691569] [25246:2371579] CHIP: [EM] Received message of type 0x31 with protocolId (0, 0) and MessageCounter:1619126971 on exchange 42877i
[1637613691569] [25246:2371579] CHIP: [EM] Rxd Ack; Removing MessageCounter:2136539305 from Retrans Table on exchange 42877i
[1637613691569] [25246:2371579] CHIP: [EM] Removed CHIP MessageCounter:2136539305 from RetransTable on exchange 42877i
[1637613691569] [25246:2371579] CHIP: [SC] Received Sigma2 msg
[1637613691569] [25246:2371579] CHIP: [SC] Peer assigned session session ID 73
[1637613691570] [25246:2371579] CHIP: [SC] Sending Sigma3
[1637613691571] [25246:2371579] CHIP: [EM] Piggybacking Ack for MessageCounter:1619126971 on exchange: 42877i
[1637613691571] [25246:2371579] CHIP: [IN] Prepared plaintext message 0x7f8dce013280 to 0x0000000000000000 of type 0x32 and protocolId (0, 0) on exchange 42877i with MessageCounter:2136539306.
[1637613691571] [25246:2371579] CHIP: [IN] Sending plaintext msg 0x7f8dce013280 with MessageCounter:2136539306 to 0x0000000000000000 at monotonic time: 428590708 msec
[1637613691571] [25246:2371579] CHIP: [SC] Sent Sigma3 msg
[1637613691579] [25246:2371581] CHIP: [EM] Received message of type 0x40 with protocolId (0, 0) and MessageCounter:1619126972 on exchange 42877i
[1637613691579] [25246:2371581] CHIP: [EM] Rxd Ack; Removing MessageCounter:2136539306 from Retrans Table on exchange 42877i
[1637613691579] [25246:2371581] CHIP: [EM] Removed CHIP MessageCounter:2136539306 from RetransTable on exchange 42877i
[1637613691579] [25246:2371581] CHIP: [SC] Success status report received. Session was established
[1637613691579] [25246:2371581] CHIP: [IN] New secure session created for device 0x0000000000000001, key 73!!
[1637613691579] [25246:2371581] CHIP: [TOO] Sending cluster (0x00000050) command (0x00000000) on endpoint 1
[1637613691579] [25246:2371581] CHIP: [DMG] ICR moving to [AddingComm]
[1637613691579] [25246:2371581] CHIP: [DMG] ICR moving to [AddedComma]
[1637613691579] [25246:2371581] CHIP: [IN] Prepared encrypted message 0x7f8dce013280 to 0x0000000000000001 of type 0x8 and protocolId (0, 1) on exchange 42878i with MessageCounter:0.
[1637613691579] [25246:2371581] CHIP: [IN] Sending encrypted msg 0x7f8dce013280 with MessageCounter:0 to 0x0000000000000001 at monotonic time: 428590717 msec
[1637613691579] [25246:2371581] CHIP: [DMG] ICR moving to [CommandSen]
[1637613691579] [25246:2371581] CHIP: [EM] Sending Standalone Ack for MessageCounter:1619126972 on exchange 42877i
[1637613691579] [25246:2371581] CHIP: [IN] Prepared plaintext message 0x700001a40450 to 0x0000000000000000 of type 0x10 and protocolId (0, 0) on exchange 42877i with MessageCounter:2136539307.
[1637613691579] [25246:2371581] CHIP: [IN] Sending plaintext msg 0x700001a40450 with MessageCounter:2136539307 to 0x0000000000000000 at monotonic time: 428590717 msec
[1637613691579] [25246:2371581] CHIP: [EM] Flushed pending ack for MessageCounter:1619126972 on exchange 42877i
[1637613691582] [25246:2371581] CHIP: [EM] Received message of type 0x9 with protocolId (0, 1) and MessageCounter:1 on exchange 42878i
[1637613691582] [25246:2371581] CHIP: [EM] Rxd Ack; Removing MessageCounter:0 from Retrans Table on exchange 42878i
[1637613691582] [25246:2371581] CHIP: [EM] Removed CHIP MessageCounter:0 from RetransTable on exchange 42878i
[1637613691582] [25246:2371581] CHIP: [DMG] InvokeCommand =
[1637613691582] [25246:2371581] CHIP: [DMG] {
[1637613691582] [25246:2371581] CHIP: [DMG] 	CommandList =
[1637613691582] [25246:2371581] CHIP: [DMG] 	[
[1637613691582] [25246:2371581] CHIP: [DMG] 		CommandDataIB =
[1637613691582] [25246:2371581] CHIP: [DMG] 		{
[1637613691582] [25246:2371581] CHIP: [DMG] 			CommandPathIB =
[1637613691582] [25246:2371581] CHIP: [DMG] 			{
[1637613691582] [25246:2371581] CHIP: [DMG] 				EndpointId = 0x1,
[1637613691582] [25246:2371581] CHIP: [DMG] 				ClusterId = 0x50,
[1637613691582] [25246:2371581] CHIP: [DMG] 				CommandId = 0x0,
[1637613691582] [25246:2371581] CHIP: [DMG] 			},
[1637613691582] [25246:2371581] CHIP: [DMG] 			
[1637613691582] [25246:2371581] CHIP: [DMG] 			StatusIB =
[1637613691582] [25246:2371581] CHIP: [DMG] 			{
[1637613691582] [25246:2371581] CHIP: [DMG] 				status = 0xc6,
[1637613691582] [25246:2371581] CHIP: [DMG] 			},
[1637613691582] [25246:2371581] CHIP: [DMG] 			
[1637613691582] [25246:2371581] CHIP: [DMG] 		},
[1637613691582] [25246:2371581] CHIP: [DMG] 		
[1637613691582] [25246:2371581] CHIP: [DMG] 	],
[1637613691582] [25246:2371581] CHIP: [DMG] 	
[1637613691582] [25246:2371581] CHIP: [DMG] }
[1637613691582] [25246:2371581] CHIP: [DMG] Received Command Response Status for Endpoint=1 Cluster=0x0000_0050 Command=0x0000_0000 Status=0xc6
[1637613691583] [25246:2371581] CHIP: [TOO] Default Failure Response: 0xc6
[1637613691583] [25246:2371581] CHIP: [DMG] ICR moving to [AwaitingDe]
[1637613691583] [25246:2371581] CHIP: [EM] Sending Standalone Ack for MessageCounter:1 on exchange 42878i
[1637613691583] [25246:2371581] CHIP: [IN] Prepared encrypted message 0x700001a40310 to 0x0000000000000001 of type 0x10 and protocolId (0, 0) on exchange 42878i with MessageCounter:1.
[1637613691583] [25246:2371581] CHIP: [IN] Sending encrypted msg 0x700001a40310 with MessageCounter:1 to 0x0000000000000001 at monotonic time: 428590721 msec
[1637613691583] [25246:2371581] CHIP: [EM] Flushed pending ack for MessageCounter:1 on exchange 42878i
[1637613691584] [25246:2371575] CHIP: [-] ../../zzz_generated/chip-tool/zap-generated/cluster/Commands.h:45: CHIP Error 0x000000AC: Internal error at ../../examples/chip-tool/commands/common/CHIPCommand.cpp:84
[1637613691584] [25246:2371575] CHIP: [TOO] Run command failure: ../../zzz_generated/chip-tool/zap-generated/cluster/Commands.h:45: CHIP Error 0x000000AC: Internal error
[1637613691618] [25246:2371575] CHIP: [CTL] Shutting down the System State, this will teardown the CHIP Stack
[1637613691618] [25246:2371575] CHIP: [DL] Inet Layer shutdown
[1637613691618] [25246:2371575] CHIP: [DL] BLE layer shutdown
[1637613691618] [25246:2371575] CHIP: [DL] System Layer shutdown
"

Garik098, 2022-01-24:
In the recent TE7.5 testing, we are seeing error code 0x87 instead of 0x85

Survensa, 2022-03-29:
For referance still facing the isssue : 

> ubuntu@matter-7:~/cntrl/connectedhomeip/examples/chip-tool/out/debug$ ./chip-tool modeselect change-to-mode 2 1 1
[1648556998.071675][19550:19550] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1648556998.072229][19550:19550] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1648556998.072387][19550:19550] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1648556998.072501][19550:19550] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1648556998.072881][19550:19550] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-t3LZ9q)
[1648556998.073532][19550:19550] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1648556998.073587][19550:19550] CHIP:DL: NVS set: chip-counters/reboot-count = 27 (0x1B)
[1648556998.074507][19550:19550] CHIP:DL: Got Ethernet interface: eth0
[1648556998.075201][19550:19550] CHIP:DL: Found the primary Ethernet interface:eth0
[1648556998.075876][19550:19550] CHIP:DL: Got WiFi interface: wlan0
[1648556998.075954][19550:19550] CHIP:DL: Failed to reset WiFi statistic counts
[1648556998.076046][19550:19550] CHIP:IN: UDP::Init bind&listen port=0
[1648556998.076180][19550:19550] CHIP:IN: UDP::Init bound to port=36387
[1648556998.076207][19550:19550] CHIP:IN: BLEBase::Init - setting/overriding transport
[1648556998.076228][19550:19550] CHIP:IN: TransportMgr initialized
[1648556998.076262][19550:19550] CHIP:DIS: Init fabric pairing table with server storage
[1648556998.076398][19550:19550] CHIP:IN: Loading from storage for fabric index 1
[1648556998.076927][19550:19550] CHIP:IN: Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000001 and root pubkey
[1648556998.076974][19550:19550] CHIP:IN: 0x04, 0x9a, 0x8e, 0xfa, 0xe6, 0x8a, 0x73, 0x21,
[1648556998.077007][19550:19550] CHIP:IN: 0x5d, 0x29, 0xc7, 0x1b, 0x97, 0xb7, 0x39, 0x8b,
[1648556998.077039][19550:19550] CHIP:IN: 0x02, 0x03, 0x69, 0xba, 0x1d, 0x67, 0x3d, 0x92,
[1648556998.077070][19550:19550] CHIP:IN: 0x3e, 0x36, 0xc2, 0xd3, 0x16, 0xd2, 0xb2, 0x0c,
[1648556998.077101][19550:19550] CHIP:IN: 0x82, 0xf8, 0x96, 0xf7, 0xa0, 0x61, 0x26, 0x4a,
[1648556998.077132][19550:19550] CHIP:IN: 0x81, 0x4e, 0x85, 0xba, 0xb8, 0x8b, 0xf9, 0xe1,
[1648556998.077162][19550:19550] CHIP:IN: 0x13, 0x23, 0xfb, 0x59, 0xf1, 0x6f, 0x9a, 0x9d,
[1648556998.077193][19550:19550] CHIP:IN: 0xe4, 0x68, 0x0c, 0xb4, 0x2c, 0x4f, 0x31, 0x63,
[1648556998.077219][19550:19550] CHIP:IN: 0xa1,
[1648556998.077341][19550:19550] CHIP:IN: Generated compressed fabric ID
[1648556998.077379][19550:19550] CHIP:IN: 0x14, 0x76, 0xe3, 0x40, 0xdf, 0xf9, 0x07, 0xd9,
[1648556998.080197][19550:19550] CHIP:IN: Loading from storage for fabric index 2
[1648556998.080591][19550:19550] CHIP:IN: Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000002 and root pubkey
[1648556998.080634][19550:19550] CHIP:IN: 0x04, 0x9a, 0x8e, 0xfa, 0xe6, 0x8a, 0x73, 0x21,
[1648556998.080666][19550:19550] CHIP:IN: 0x5d, 0x29, 0xc7, 0x1b, 0x97, 0xb7, 0x39, 0x8b,
[1648556998.080722][19550:19550] CHIP:IN: 0x02, 0x03, 0x69, 0xba, 0x1d, 0x67, 0x3d, 0x92,
[1648556998.080753][19550:19550] CHIP:IN: 0x3e, 0x36, 0xc2, 0xd3, 0x16, 0xd2, 0xb2, 0x0c,
[1648556998.080783][19550:19550] CHIP:IN: 0x82, 0xf8, 0x96, 0xf7, 0xa0, 0x61, 0x26, 0x4a,
[1648556998.080813][19550:19550] CHIP:IN: 0x81, 0x4e, 0x85, 0xba, 0xb8, 0x8b, 0xf9, 0xe1,
[1648556998.080843][19550:19550] CHIP:IN: 0x13, 0x23, 0xfb, 0x59, 0xf1, 0x6f, 0x9a, 0x9d,
[1648556998.080873][19550:19550] CHIP:IN: 0xe4, 0x68, 0x0c, 0xb4, 0x2c, 0x4f, 0x31, 0x63,
[1648556998.080900][19550:19550] CHIP:IN: 0xa1,
[1648556998.080967][19550:19550] CHIP:IN: Generated compressed fabric ID
[1648556998.081003][19550:19550] CHIP:IN: 0x2c, 0x56, 0x73, 0xec, 0x92, 0x91, 0x75, 0x94,
[1648556998.081359][19550:19550] CHIP:IN: Loading from storage for fabric index 3
[1648556998.081709][19550:19550] CHIP:IN: Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000003 and root pubkey
[1648556998.081752][19550:19550] CHIP:IN: 0x04, 0x9a, 0x8e, 0xfa, 0xe6, 0x8a, 0x73, 0x21,
[1648556998.081783][19550:19550] CHIP:IN: 0x5d, 0x29, 0xc7, 0x1b, 0x97, 0xb7, 0x39, 0x8b,
[1648556998.081814][19550:19550] CHIP:IN: 0x02, 0x03, 0x69, 0xba, 0x1d, 0x67, 0x3d, 0x92,
[1648556998.081845][19550:19550] CHIP:IN: 0x3e, 0x36, 0xc2, 0xd3, 0x16, 0xd2, 0xb2, 0x0c,
[1648556998.081875][19550:19550] CHIP:IN: 0x82, 0xf8, 0x96, 0xf7, 0xa0, 0x61, 0x26, 0x4a,
[1648556998.081905][19550:19550] CHIP:IN: 0x81, 0x4e, 0x85, 0xba, 0xb8, 0x8b, 0xf9, 0xe1,
[1648556998.081936][19550:19550] CHIP:IN: 0x13, 0x23, 0xfb, 0x59, 0xf1, 0x6f, 0x9a, 0x9d,
[1648556998.081966][19550:19550] CHIP:IN: 0xe4, 0x68, 0x0c, 0xb4, 0x2c, 0x4f, 0x31, 0x63,
[1648556998.081993][19550:19550] CHIP:IN: 0xa1,
[1648556998.082047][19550:19550] CHIP:IN: Generated compressed fabric ID
[1648556998.082082][19550:19550] CHIP:IN: 0x5b, 0xf4, 0xa2, 0x9a, 0x0b, 0x0b, 0xfe, 0xea,
[1648556998.082441][19550:19550] CHIP:DIS: Add fabric pairing table delegate
[1648556998.082730][19550:19550] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-On9lPE)
[1648556998.083501][19550:19550] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1648556998.083566][19550:19550] CHIP:DL: NVS set: chip-counters/GlobalMCTR = 27000 (0x6978)
[1648556998.086924][19550:19550] CHIP:ZCL: Using ZAP configuration...
[... cut ...]
[1648556998.372279][19550:19550] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-NctEh0)
[1648556998.373274][19550:19550] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1648556998.373363][19550:19550] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1648556998.373401][19550:19550] CHIP:DL: Inet Layer shutdown
[1648556998.373434][19550:19550] CHIP:DL: BLE shutdown
[1648556998.373468][19550:19550] CHIP:DL: System Layer shutdown
[1648556998.373628][19550:19550] CHIP:TOO: Run command failure: IM Error 0x00000587: General error: 0x87

andy31415, 2022-04-14:
@Garik098 - please describe spec requirement (unclear to me if ChangeToMode is M(andatory) or O(ptional) in the spec from just reading the bug description) and the test case in a condensed format.

Logs in this bug are too long to parse easily. I need:
- what spec are we testing
- what is being done to test
- what environment was this tested on (what app like all-clusters/lock/light etc. and what platform(s) like esp32/nrf/efr/...) for a developer to reproduce the issue

Beyond that logs either as attachments or if reviewed by a human, condensed to minimal data (e.g. saying `Error 0xAC` or `Error 0x87` rather than a full log dump)

somu1710, 2022-04-21:
Problem:
  while executing testcase [TC-MOD-2.1] expected outcome is not matched with the result.

command used:
./chip-tool modeselect change-to-mode 2 1 1

Expected outcome:
INVALID_COMMAND (hex value 0x85/decimal 133)

Actual outcome:
 Error: IM Error 0x00000587: General error: 0x87

Reason:
Because we execute the support mode  attribute ,It should display the supporting modes values as 0,4 and 7.We are changing the mode which is not displayed in the supported modes, it should get a value as (0x85-invalid command) but actually we are getting (0x87-General error).

Modes which are supports are displayed below:

./chip-tool modeselect read supported-modes 1 1

  SupportedModes: 3 entries
     [1]: {
       Label: Black
      Mode: 0
       SemanticTag: 0
     }
:     [2]: {
      Label: Cappuccino
     Mode: 4
    SemanticTag: 0
      }
    [3]: {
      Label: Espresso
   Mode: 7
     SemanticTag: 0
    }

bzbarsky-apple, 2022-05-05:
Fixed in #17685

closed 2022-05-05 by bzbarsky-apple