project-chip/connectedhomeip #12130: [TE7]chip-tool-ipv6 fails to resolve address on ESP32 device

Kxuan, 2021-11-23:
#### Problem
Unable to pairing wifi device with chip-tool-ipv6.
 
The avahi resolve fail with an "Invalid address" error.

```
[1637321134.947233][148429:148434] CHIP:DL: Avahi resolve found
[1637321134.947324][148429:148434] CHIP:DIS: Ignoring IPv4 mDNS address.
[1637321134.947405][148429:148434] CHIP:DIS: Node ID resolved failed with ../../third_party/connectedhomeip/src/platform/Linux/DnssdImpl.cpp:727: CHIP Error 0x0000002E: Invalid address
[1637321134.947472][148429:148434] CHIP:CTL: Error resolving node id: ../../third_party/connectedhomeip/src/platform/Linux/DnssdImpl.cpp:727: CHIP Error 0x0000002E: Invalid address
[1637321134.947533][148429:148434] CHIP:TOO: OnAddressUpdateComplete: 0: ../../third_party/connectedhomeip/src/platform/Linux/DnssdImpl.cpp:727: CHIP Error 0x0000002E: Invalid address
```

RaspberryPI (TH) log:[th.log](https://github.com/project-chip/connectedhomeip/files/7585472/th.log)
ESP32 (DUT) log: [dut.log](https://github.com/project-chip/connectedhomeip/files/7585474/dut.log)

Kxuan, 2021-11-23:
By setting `CONFIG_IPV4_DISABLED` to `y` in ESP32, there is a new error message:

TH:
```
[1637634630.666169][16034:16039] CHIP:SC: CASESession timed out while waiting for a response from the peer. Current state was 1
[1637634630.666283][16034:16039] CHIP:CTL: Device connection failed. Error ../../third_party/connectedhomeip/src/protocols/secure_channel/CASESession.cpp:291: CHIP Error 0x00000032: Timeout
[1637634630.666352][16034:16039] CHIP:TOO: Device commissioning Failure: ../../third_party/connectedhomeip/src/protocols/secure_channel/CASESession.cpp:291: CHIP Error 0x00000032: Timeout
[1637634630.667101][16034:16034] CHIP:-: ../../third_party/connectedhomeip/src/protocols/secure_channel/CASESession.cpp:291: CHIP Error 0x00000032: Timeout at ../../commands/common/CHIPCommand.cpp:84
[1637634630.667218][16034:16034] CHIP:TOO: Run command failure: ../../third_party/connectedhomeip/src/protocols/secure_channel/CASESession.cpp:291: CHIP Error 0x00000032: Timeout
```

RaspberryPI (TH) log: [th.log](https://github.com/project-chip/connectedhomeip/files/7585518/th.log)
ESP32 (DUT) log: [dut.log](https://github.com/project-chip/connectedhomeip/files/7585517/dut.log)

dhrishi, 2022-01-18:
@Kxuan Can you please retry on the latest codebase. The chip-tool now successful resolves IPv6 address on ESP32

Kxuan, 2022-01-20:
@dhrishi I am currently working on TE7.5, using commit b0af6ba42b6ce494748f18ecf3e054a400b8f45b .

I can confirm that the chip-tool can resolve the ESP32's IPv6 address and establish CASE session over IPv6.

dhrishi, 2022-01-27:
Closing it, as it seems to work fine from the above comment

closed 2022-01-27 by dhrishi