project-chip/connectedhomeip #12131: Crash on contentlauncher launch-content linux tv

xylophone21, 2021-11-23:
#### Problem
* code: 2e16d32966a30146235fca6b873289e2765671d3
* contoller macos M1
   * ./scripts/examples/gn_build_example.sh examples/chip-tool out/chip-tool
   * ./out/chip-tool/chip-tool pairing ethernet 112244 20202021 3840 192.168.88.251 5540
   * ./out/chip-tool/chip-tool contentlauncher launch-content 1 abc 112244 
* tvapp on Raspberry Pi 4 
    * ./scripts/examples/gn_build_example.sh examples/tv-app/linux/ out/host/
    * sudo ./out/host/chip-tv-app
* Segmentation fault on side

[RaspberryPi4.log](https://github.com/project-chip/connectedhomeip/files/7585590/RaspberryPi4.log)
[RaspberryPi4.log](https://github.com/project-chip/connectedhomeip/files/7585591/RaspberryPi4.log)

xylophone21, 2021-11-24:
Seems it is because responses can not use struct as param. All responses using struct are in todo.

like: src/app/clusters/tv-channel-server/tv-channel-server.cpp

    std::string matchString(match.data(), match.size());
    // TODO: Enable this once struct as param is supported
    // TvChannelInfo channelInfo = tvChannelClusterChangeChannel(matchString);
    // sendResponse(command, channelInfo);
    emberAfSendImmediateDefaultResponse(EMBER_ZCL_STATUS_FAILURE);

xylophone21, 2021-11-27:
fixed at

https://github.com/project-chip/connectedhomeip/pull/12250

lazarkov, 2022-01-23:
Fixed

closed 2022-01-23 by lazarkov