project-chip/connectedhomeip #12132: Internal error on contentlauncher launch-content chip-all-clusters-app

xylophone21, 2021-11-23:
#### Problem
 * code: 2e16d32
 * contoller macos M1
     *  ./scripts/examples/gn_build_example.sh examples/chip-tool out/chip-tool
     * ./out/chip-tool/chip-tool pairing ethernet 112244 20202021 3840 192.168.88.251 5540
     * ./out/chip-tool/chip-tool contentlauncher launch-content 1 abc 112244
 * chip-all-clusters-app on Raspberry Pi 4
     * ./scripts/examples/gn_build_example.sh examples/all-clusters-app/linux/ out/all
     *  sudo out/all/chip-all-clusters-app
 
 `[1637636012895] [86480:4937482] CHIP: [EM] Removed CHIP MessageCounter:1 from RetransTable on exchange 5790i
[1637636012895] [86480:4937482] CHIP: [DMG] InvokeResponseMessage =
[1637636012895] [86480:4937482] CHIP: [DMG] {
[1637636012895] [86480:4937482] CHIP: [DMG]     suppressResponse = false, 
[1637636012895] [86480:4937482] CHIP: [DMG]     InvokeResponses =
[1637636012895] [86480:4937482] CHIP: [DMG]     [
[1637636012895] [86480:4937482] CHIP: [DMG]             InvokeResponseIB =
[1637636012895] [86480:4937482] CHIP: [DMG]             {
[1637636012896] [86480:4937482] CHIP: [DMG]                     CommandStatusIB =
[1637636012896] [86480:4937482] CHIP: [DMG]                     {
[1637636012896] [86480:4937482] CHIP: [DMG]                             CommandPathIB =
[1637636012896] [86480:4937482] CHIP: [DMG]                             {
[1637636012896] [86480:4937482] CHIP: [DMG]                                     EndpointId = 0x1,
[1637636012896] [86480:4937482] CHIP: [DMG]                                     ClusterId = 0x50a,
[1637636012896] [86480:4937482] CHIP: [DMG]                                     CommandId = 0x0,
[1637636012896] [86480:4937482] CHIP: [DMG]                             },
[1637636012896] [86480:4937482] CHIP: [DMG] 
[1637636012896] [86480:4937482] CHIP: [DMG]                             StatusIB =
[1637636012896] [86480:4937482] CHIP: [DMG]                             {
[1637636012896] [86480:4937482] CHIP: [DMG]                                     status = 0xc3,
[1637636012896] [86480:4937482] CHIP: [DMG]                             },
[1637636012896] [86480:4937482] CHIP: [DMG] 
[1637636012902] [86480:4937482] CHIP: [DMG]                     },
[1637636012902] [86480:4937482] CHIP: [DMG] 
[1637636012902] [86480:4937482] CHIP: [DMG]             },
[1637636012902] [86480:4937482] CHIP: [DMG] 
[1637636012902] [86480:4937482] CHIP: [DMG]     ],
[1637636012902] [86480:4937482] CHIP: [DMG] 
[1637636012902] [86480:4937482] CHIP: [DMG] },
[1637636012902] [86480:4937482] CHIP: [DMG] Received Command Response Status for Endpoint=1 Cluster=0x0000_050A Command=0x0000_0000 Status=0xc3
[1637636012902] [86480:4937482] CHIP: [TOO] Default Failure Response: 0xc3
[1637636012902] [86480:4937482] CHIP: [DMG] ICR moving to [AwaitingDe]
[1637636012902] [86480:4937482] CHIP: [EM] Sending Standalone Ack for MessageCounter:1 on exchange 5790i
[1637636012902] [86480:4937482] CHIP: [IN] Prepared encrypted message 0x16d3b1108 to 0x000000000001B674 of type 0x10 and protocolId (0, 0) on exchange 5790i with MessageCounter:2.
[1637636012915] [86480:4937482] CHIP: [IN] Sending encrypted msg 0x16d3b1108 with MessageCounter:2 to 0x000000000001B674 at monotonic time: 692391017 msec
[1637636012915] [86480:4937482] CHIP: [EM] Flushed pending ack for MessageCounter:1 on exchange 5790i
[1637636012915] [86480:4937478] CHIP: [-] ../../examples/chip-tool/third_party/connectedhomeip/zzz_generated/chip-tool/zap-generated/cluster/Commands.h:2054: CHIP Error 0x000000AC: Internal error at ../../examples/chip-tool/commands/common/CHIPCommand.cpp
[1637636012915] [86480:4937478] CHIP: [TOO] Run command failure: ../../examples/chip-tool/third_party/connectedhomeip/zzz_generated/chip-tool/zap-generated/cluster/Commands.h:2054: CHIP Error 0x000000AC: Internal error
[1637636012920] [86480:4937478] CHIP: [CTL] Shutting down the System State, this will teardown the CHIP Stack
[1637636012920] [86480:4937478] CHIP: [DL] Inet Layer shutdown
[1637636012920] [86480:4937478] CHIP: [DL] BLE shutdown
[1637636012920] [86480:4937478] CHIP: [DL] System Layer shutdown`

Seems the CommandId is wrong, it should be 3 | Launcher Response?

[RaspberryPi41.log](https://github.com/project-chip/connectedhomeip/files/7585608/RaspberryPi41.log)
[MacBook-Air1.log](https://github.com/project-chip/connectedhomeip/files/7585607/MacBook-Air1.log)

yufengwangca, 2022-02-07:
@lazarkov 

Could you help to take a look and confirm if the test chip-tool command is out of date?

woody-apple, 2022-02-08:
This was just added in a merged PR today I believe, can you please confirm?

closed 2022-02-08 by woody-apple