project-chip/connectedhomeip #12133: TC-MF-1.20 Remove fabric attribute is removing the fabric but is giving internal error response

Garik098, 2021-11-23:
In Multi Fabric testing, Remove Fabric attribute is removing the fabric correctly but is giving internal error response. 

In the below logs, I read the fabric list and it displayed 3 fabrics. In the second log I removed 2nd fabric and it gave internal error response. In the third log, I read the fabric list again and I see that the fabric 2 has been removed. 

adk@adks-Mac-mini debug % ./chip-tool operationalcredentials read fabrics-list 1 1        
[1637636342098] [27045:2533138] CHIP: [CTL] Generating NOC
[1637636342098] [27045:2533138] CHIP: [CTL] Generating ICAC
[1637636342098] [27045:2533138] CHIP: [DL] AUDIT: ===== RANDOM NUMBER GENERATOR AUDIT START ====
[1637636342098] [27045:2533138] CHIP: [DL] AUDIT: * Validate buf1 and buf2 are <<<different every run/boot!>>>
[1637636342098] [27045:2533138] CHIP: [DL] AUDIT: * Validate r1 and r2 are <<<different every run/boot!>>>
[1637636342098] [27045:2533138] CHIP: [DL] AUDIT: * buf1: FD9B151C26498F35EC74BBB45B99490C
[1637636342098] [27045:2533138] CHIP: [DL] AUDIT: * buf2: 75C314347E1186FA02293006F12B3E96
[1637636342098] [27045:2533138] CHIP: [DL] AUDIT: * r1: 0x22E413A9 r2: 0x051F6829
[1637636342098] [27045:2533138] CHIP: [DL] AUDIT: ===== RANDOM NUMBER GENERATOR AUDIT END ====
[1637636342098] [27045:2533138] CHIP: [DL] _Init
[1637636342098] [27045:2533138] CHIP: [IN] UDP::Init bind&listen port=5541
[1637636342099] [27045:2533138] CHIP: [IN] UDP::Init bound to port=5541
[1637636342099] [27045:2533138] CHIP: [IN] UDP::Init bind&listen port=5541
[1637636342099] [27045:2533138] CHIP: [IN] UDP::Init bound to port=5541
[1637636342099] [27045:2533138] CHIP: [IN] TransportMgr initialized
[1637636342099] [27045:2533138] CHIP: [DIS] Init fabric pairing table with server storage
[1637636342099] [27045:2533138] CHIP: [CTL] System State Initialized...
[1637636342099] [27045:2533138] CHIP: [ZCL] Using ZAP configuration...
[1637636342099] [27045:2533138] CHIP: [DIS] Verifying the received credentials
[1637636342100] [27045:2533138] CHIP: [DIS] Added new fabric at index: 1, Initialized: 1
[1637636342100] [27045:2533138] CHIP: [DIS] Assigned compressed fabric ID: 0x56791D9DD63C9A95, node ID: 0x000000000001B669
[1637636342100] [27045:2533138] CHIP: [CTL] Joined the fabric at index 1. Compressed fabric ID is: 0x56791D9DD63C9A95
[1637636342100] [27045:2533144] CHIP: [DL] _OnPlatformEvent
[1637636342100] [27045:2533144] CHIP: [TOO] Sending command to node 0x1
[1637636342101] [27045:2533144] CHIP: [IN] Prepared plaintext message 0x7fdfb7808280 to 0x0000000000000000 of type 0x30 and protocolId (0, 0) on exchange 28601i with MessageCounter:692019230.
[1637636342101] [27045:2533144] CHIP: [IN] Sending plaintext msg 0x7fdfb7808280 with MessageCounter:692019230 to 0x0000000000000000 at monotonic time: 451240492 msec
[1637636342101] [27045:2533144] CHIP: [SC] Sent Sigma1 msg
[1637636342114] [27045:2533144] CHIP: [EM] Received message of type 0x31 with protocolId (0, 0) and MessageCounter:1575357335 on exchange 28601i
[1637636342114] [27045:2533144] CHIP: [EM] Rxd Ack; Removing MessageCounter:692019230 from Retrans Table on exchange 28601i
[1637636342114] [27045:2533144] CHIP: [EM] Removed CHIP MessageCounter:692019230 from RetransTable on exchange 28601i
[1637636342114] [27045:2533144] CHIP: [SC] Received Sigma2 msg
[1637636342114] [27045:2533144] CHIP: [SC] Peer assigned session session ID 9
[1637636342116] [27045:2533144] CHIP: [SC] Sending Sigma3
[1637636342116] [27045:2533144] CHIP: [EM] Piggybacking Ack for MessageCounter:1575357335 on exchange: 28601i
[1637636342116] [27045:2533144] CHIP: [IN] Prepared plaintext message 0x7fdfb7808280 to 0x0000000000000000 of type 0x32 and protocolId (0, 0) on exchange 28601i with MessageCounter:692019231.
[1637636342116] [27045:2533144] CHIP: [IN] Sending plaintext msg 0x7fdfb7808280 with MessageCounter:692019231 to 0x0000000000000000 at monotonic time: 451240507 msec
[1637636342116] [27045:2533144] CHIP: [SC] Sent Sigma3 msg
[1637636342124] [27045:2533144] CHIP: [EM] Received message of type 0x40 with protocolId (0, 0) and MessageCounter:1575357336 on exchange 28601i
[1637636342124] [27045:2533144] CHIP: [EM] Rxd Ack; Removing MessageCounter:692019231 from Retrans Table on exchange 28601i
[1637636342124] [27045:2533144] CHIP: [EM] Removed CHIP MessageCounter:692019231 from RetransTable on exchange 28601i
[1637636342124] [27045:2533144] CHIP: [SC] Success status report received. Session was established
[1637636342124] [27045:2533144] CHIP: [IN] New secure session created for device 0x0000000000000001, key 9!!
[1637636342124] [27045:2533144] CHIP: [TOO] Sending cluster (0x003E) command (0x00) on endpoint 1
[1637636342124] [27045:2533144] CHIP: [DMG] SendReadRequest: Client[0] [ INIT]
[1637636342124] [27045:2533144] CHIP: [IN] Prepared encrypted message 0x7fdfb7808280 to 0x0000000000000001 of type 0x2 and protocolId (0, 1) on exchange 28602i with MessageCounter:0.
[1637636342124] [27045:2533144] CHIP: [IN] Sending encrypted msg 0x7fdfb7808280 with MessageCounter:0 to 0x0000000000000001 at monotonic time: 451240516 msec
[1637636342124] [27045:2533144] CHIP: [DMG] Client[0] moving to [AwaitingInitialReport]
[1637636342124] [27045:2533144] CHIP: [EM] Sending Standalone Ack for MessageCounter:1575357336 on exchange 28601i
[1637636342124] [27045:2533144] CHIP: [IN] Prepared plaintext message 0x70000e8dc450 to 0x0000000000000000 of type 0x10 and protocolId (0, 0) on exchange 28601i with MessageCounter:692019232.
[1637636342124] [27045:2533144] CHIP: [IN] Sending plaintext msg 0x70000e8dc450 with MessageCounter:692019232 to 0x0000000000000000 at monotonic time: 451240516 msec
[1637636342124] [27045:2533144] CHIP: [EM] Flushed pending ack for MessageCounter:1575357336 on exchange 28601i
[1637636342131] [27045:2533144] CHIP: [EM] Received message of type 0x5 with protocolId (0, 1) and MessageCounter:1 on exchange 28602i
[1637636342131] [27045:2533144] CHIP: [EM] Rxd Ack; Removing MessageCounter:0 from Retrans Table on exchange 28602i
[1637636342131] [27045:2533144] CHIP: [EM] Removed CHIP MessageCounter:0 from RetransTable on exchange 28602i
[1637636342131] [27045:2533144] CHIP: [DMG] ReportData =
[1637636342131] [27045:2533144] CHIP: [DMG] {
[1637636342131] [27045:2533144] CHIP: [DMG] 	AttributeDataList =
[1637636342131] [27045:2533144] CHIP: [DMG] 	[
[1637636342131] [27045:2533144] CHIP: [DMG] 		AttributeDataElement =
[1637636342131] [27045:2533144] CHIP: [DMG] 		{
[1637636342131] [27045:2533144] CHIP: [DMG] 			AttributePath =
[1637636342131] [27045:2533144] CHIP: [DMG] 			{
[1637636342131] [27045:2533144] CHIP: [DMG] 				NodeId = 0x1,
[1637636342131] [27045:2533144] CHIP: [DMG] 				EndpointId = 0x1,
[1637636342131] [27045:2533144] CHIP: [DMG] 				ClusterId = 0x3e,
[... 277 lines cut ...]
[1637636385264] [27126:2533645] CHIP: [TOO] FabricDescriptor[1]:
[1637636385264] [27126:2533645] CHIP: [TOO]   fabricIndex: 1
[1637636385264] [27126:2533645] CHIP: [ZCL]   RootPublicKey: 65
[1637636385264] [27126:2533645] CHIP: [TOO]   vendorId: 65521
[1637636385264] [27126:2533645] CHIP: [TOO]   fabricId: 0
[1637636385264] [27126:2533645] CHIP: [TOO]   nodeId: 1
[1637636385264] [27126:2533645] CHIP: [ZCL]   Label: 
[1637636385264] [27126:2533645] CHIP: [TOO] FabricDescriptor[2]:
[1637636385264] [27126:2533645] CHIP: [TOO]   fabricIndex: 3
[1637636385264] [27126:2533645] CHIP: [ZCL]   RootPublicKey: 65
[1637636385264] [27126:2533645] CHIP: [TOO]   vendorId: 65521
[1637636385264] [27126:2533645] CHIP: [TOO]   fabricId: 0
[1637636385264] [27126:2533645] CHIP: [TOO]   nodeId: 1
[1637636385264] [27126:2533645] CHIP: [ZCL]   Label: 
[1637636385264] [27126:2533645] CHIP: [EM] Piggybacking Ack for MessageCounter:1 on exchange: 12531i
[1637636385264] [27126:2533645] CHIP: [IN] Prepared encrypted message 0x7feeab80a680 to 0x0000000000000001 of type 0x1 and protocolId (0, 1) on exchange 12531i with MessageCounter:1.
[1637636385264] [27126:2533645] CHIP: [IN] Sending encrypted msg 0x7feeab80a680 with MessageCounter:1 to 0x0000000000000001 at monotonic time: 451283656 msec
[1637636385265] [27126:2533645] CHIP: [DMG] Client[0] moving to [UNINIT]
[1637636385265] [27126:2533641] CHIP: [CTL] Shutting down the commissioner
[1637636385265] [27126:2533641] CHIP: [CTL] Shutting down the controller
[1637636385266] [27126:2533641] CHIP: [CTL] Shutting down the System State, this will teardown the CHIP Stack
[1637636385266] [27126:2533641] CHIP: [DL] Inet Layer shutdown
[1637636385266] [27126:2533641] CHIP: [DL] BLE layer shutdown
[1637636385266] [27126:2533641] CHIP: [DL] System Layer shutdown
adk@adks-Mac-mini debug %

holbrookt, 2022-01-31:
@Garik098 is this still an issue?

sowmyassp, 2022-02-07:
@holbrookt Now we are not facing this issue.

closed 2022-02-07 by holbrookt