project-chip/connectedhomeip #12137: TC-MF Mac chip tool testing: Getting segmentation fault when doing few sequences

Garik098, 2021-11-23:
As a part of Multi Fabric testing, while switching between ECM and BCM or doing the 20 wrong passcode sequence test cases, I am hitting the segmentation fault issue. 

The segmentation fault issue is being seen both on chip tool log as well as on the app (I checked on both lighting and all cluster apps and it is happening on both). Below are the chip tool and accessory side logs,

Chip tool log: 
ubuntu@matter-te-6:~/TE7/connectedhomeip/out/debug$ ./chip-tool pairing onnetwork 1 20202025
[1637639678.061508][3620:3620] CHIP:CTL: Generating NOC
[1637639678.061926][3620:3620] CHIP:CTL: Generating ICAC
[1637639678.067101][3620:3620] CHIP:DL: AUDIT: ===== RANDOM NUMBER GENERATOR AUDIT START ====
[1637639678.067147][3620:3620] CHIP:DL: AUDIT: * Validate buf1 and buf2 are <<<different every run/boot!>>>
[1637639678.067174][3620:3620] CHIP:DL: AUDIT: * Validate r1 and r2 are <<<different every run/boot!>>>
[1637639678.067198][3620:3620] CHIP:DL: AUDIT: * buf1: 73C8A3028EAEB53F2429B66C17F56677
[1637639678.067221][3620:3620] CHIP:DL: AUDIT: * buf2: DA10C2E301D48F99374D6AD6830D5EB9
[1637639678.067287][3620:3620] CHIP:DL: AUDIT: * r1: 0x7AA7C1AC r2: 0x1DFE11B9
[1637639678.067310][3620:3620] CHIP:DL: AUDIT: ===== RANDOM NUMBER GENERATOR AUDIT END ====
[1637639678.067896][3620:3620] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-hwgyO2)
[1637639678.068494][3620:3620] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1637639678.068546][3620:3620] CHIP:DL: NVS set: chip-counters/reboot-count = 15 (0xF)
[1637639678.069553][3620:3620] CHIP:DL: Got Ethernet interface: eth0
[1637639678.070332][3620:3620] CHIP:DL: Found the primary Ethernet interface:eth0
[1637639678.071066][3620:3620] CHIP:DL: Got WiFi interface: wlan0
[1637639678.077299][3620:3620] CHIP:DL: Found the primary WiFi interface:wlan0
[1637639678.077368][3620:3620] CHIP:IN: UDP::Init bind&listen port=5541
[1637639678.077491][3620:3620] CHIP:IN: UDP::Init bound to port=5541
[1637639678.077521][3620:3620] CHIP:IN: UDP::Init bind&listen port=5541
[1637639678.077627][3620:3620] CHIP:IN: UDP::Init bound to port=5541
[1637639678.077655][3620:3620] CHIP:IN: TransportMgr initialized
[1637639678.077724][3620:3620] CHIP:DIS: Init fabric pairing table with server storage
[1637639678.078111][3620:3620] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-dRPSq5)
[1637639678.078709][3620:3620] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1637639678.078764][3620:3620] CHIP:DL: NVS set: chip-counters/GlobalMCTR = 15000 (0x3A98)
[1637639678.078902][3620:3620] CHIP:CTL: System State Initialized...
[1637639678.084666][3620:3620] CHIP:DL: Avahi client registered
[1637639678.086379][3620:3620] CHIP:ZCL: Using ZAP configuration...
[1637639678.087234][3620:3620] CHIP:DIS: Verifying the received credentials
[1637639678.089968][3620:3620] CHIP:DIS: Added new fabric at index: 1, Initialized: 1
[1637639678.090015][3620:3620] CHIP:DIS: Assigned compressed fabric ID: 0xC30C7FDDB86679A6, node ID: 0x000000000001B669
[1637639678.090050][3620:3620] CHIP:CTL: Joined the fabric at index 1. Compressed fabric ID is: 0xC30C7FDDB86679A6
[1637639678.090474][3620:3625] CHIP:DL: CHIP task running
[1637639678.090715][3620:3625] CHIP:DL: HandlePlatformSpecificBLEEvent 32785
[1637639678.102528][3620:3625] CHIP:DL: Avahi browse: cache new
[1637639678.102801][3620:3625] CHIP:DL: Avahi browse: cache new
[1637639678.102931][3620:3625] CHIP:DL: Avahi browse: cache new
[1637639678.103137][3620:3625] CHIP:DL: Avahi browse: cache new
[1637639678.103230][3620:3625] CHIP:DL: Avahi browse: cache new
[1637639678.103408][3620:3625] CHIP:DL: Avahi browse: cache new
[1637639678.103611][3620:3625] CHIP:DL: Avahi browse: cache exhausted
[1637639679.106045][3620:3625] CHIP:DL: Avahi browse: all for now
[1637639679.212893][3620:3625] CHIP:DL: Avahi resolve found
[1637639679.213067][3620:3625] CHIP:TOO: Discovered Device: fe80::e65f:1ff:fe2e:ed54:5540
[1637639679.213609][3620:3625] CHIP:SC: Assigned local session key ID 25
[1637639679.213872][3620:3625] CHIP:IN: Prepared plaintext message 0xaaaae59c4e30 to 0x0000000000000000 of type 0x20 and protocolId (0, 0) on exchange 49567i with MessageCounter:3635204436.
[1637639679.213975][3620:3625] CHIP:IN: Sending plaintext msg 0xaaaae59c4e30 with MessageCounter:3635204436 to 0x0000000000000000 at monotonic time: 4874536 msec
[1637639679.214250][3620:3625] CHIP:SC: Sent PBKDF param request
[1637639679.218057][3620:3625] CHIP:EM: Received message of type 0x21 with protocolId (0, 0) and MessageCounter:442006820 on exchange 49567i
[1637639679.218191][3620:3625] CHIP:EM: Rxd Ack; Removing MessageCounter:3635204436 from Retrans Table on exchange 49567i
[1637639679.218255][3620:3625] CHIP:EM: Removed CHIP MessageCounter:3635204436 from RetransTable on exchange 49567i
[1637639679.218338][3620:3625] CHIP:SC: Received PBKDF param response
[1637639679.218431][3620:3625] CHIP:SC: Peer assigned session ID 4
[1637639679.220771][3620:3625] CHIP:EM: Piggybacking Ack for MessageCounter:442006820 on exchange: 49567i
[1637639679.220832][3620:3625] CHIP:IN: Prepared plaintext message 0xaaaae59c4e30 to 0x0000000000000000 of type 0x22 and protocolId (0, 0) on exchange 49567i with MessageCounter:3635204437.
[1637639679.220886][3620:3625] CHIP:IN: Sending plaintext msg 0xaaaae59c4e30 with MessageCounter:3635204437 to 0x0000000000000000 at monotonic time: 4874543 msec
[1637639679.221045][3620:3625] CHIP:SC: Sent spake2p msg1
[1637639679.223356][3620:3625] CHIP:EM: Received message of type 0x23 with protocolId (0, 0) and MessageCounter:442006821 on exchange 49567i
[1637639679.223415][3620:3625] CHIP:EM: Rxd Ack; Removing MessageCounter:3635204437 from Retrans Table on exchange 49567i
[1637639679.223452][3620:3625] CHIP:EM: Removed CHIP MessageCounter:3635204437 from RetransTable on exchange 49567i
[1637639679.223498][3620:3625] CHIP:SC: Received spake2p msg2
[1637639679.225239][3620:3625] CHIP:SC: Failed to verify peer's MAC. This can happen when setup code is incorrect.
[1637639679.225279][3620:3625] CHIP:SC: Sending status report. Protocol code 2, exchange 49567
[1637639679.225324][3620:3625] CHIP:EM: Piggybacking Ack for MessageCounter:442006821 on exchange: 49567i
[... 70 lines cut ...]
[1637639679.217766][3692:3692] CHIP:IN: Sending plaintext msg 0xaaaadb4e9ae8 with MessageCounter:442006820 to 0x0000000000000000 at monotonic time: 4871093 msec
[1637639679.218026][3692:3692] CHIP:SC: Sent PBKDF param response
[1637639679.222056][3692:3692] CHIP:EM: Received message of type 0x22 with protocolId (0, 0) and MessageCounter:3635204437 on exchange 49567r
[1637639679.222104][3692:3692] CHIP:EM: Rxd Ack; Removing MessageCounter:442006820 from Retrans Table on exchange 49567r
[1637639679.222131][3692:3692] CHIP:EM: Removed CHIP MessageCounter:442006820 from RetransTable on exchange 49567r
[1637639679.222165][3692:3692] CHIP:SC: Received spake2p msg1
[1637639679.223942][3692:3692] CHIP:EM: Piggybacking Ack for MessageCounter:3635204437 on exchange: 49567r
[1637639679.223983][3692:3692] CHIP:IN: Prepared plaintext message 0xaaaadb4e9ae8 to 0x0000000000000000 of type 0x23 and protocolId (0, 0) on exchange 49567r with MessageCounter:442006821.
[1637639679.224020][3692:3692] CHIP:IN: Sending plaintext msg 0xaaaadb4e9ae8 with MessageCounter:442006821 to 0x0000000000000000 at monotonic time: 4871100 msec
[1637639679.224125][3692:3692] CHIP:SC: Sent spake2p msg2
[1637639679.226544][3692:3692] CHIP:EM: Received message of type 0x40 with protocolId (0, 0) and MessageCounter:3635204438 on exchange 49567r
[1637639679.226587][3692:3692] CHIP:EM: Rxd Ack; Removing MessageCounter:442006821 from Retrans Table on exchange 49567r
[1637639679.226614][3692:3692] CHIP:EM: Removed CHIP MessageCounter:442006821 from RetransTable on exchange 49567r
[1637639679.226657][3692:3692] CHIP:SC: Received error (protocol code 2) during PASE process. ../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/protocols/secure_channel/PASESession.cpp:827: CHIP Error 0x00000038: Invalid PASE parameter
[1637639679.226715][3692:3692] CHIP:SC: Failed during PASE session setup. ../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/protocols/secure_channel/PASESession.cpp:827: CHIP Error 0x00000038: Invalid PASE parameter
[1637639679.226745][3692:3692] CHIP:SVR: Commissioning failed (attempt 1): ../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/protocols/secure_channel/PASESession.cpp:827: CHIP Error  [... cut ...]

Even though it happened randomly, I am seeing this issue more clearly when I follow the following sequence,

1 -> Provision first controller and open commissioning window using BCM
2 -> Provision the second controller and open commissioning window using ECM
3 -> Provision the third controller. Observe that the third controller will show segmentation fault on its logs and the accessory too will show segmentation fault in its log and end the process.

Please do let me know if you need more info.

woody-apple, 2022-02-08:
@Garik098 I believe this is passing tests now, can you please confirm?

closed 2022-02-08 by woody-apple

lisaaaas, 2022-08-31:
I met the same problem today. Is there any solution