project-chip/connectedhomeip #12144: TC-PCC chip tool testing: writing to operation mode and control mode are not updating in effective operation mode and effective control mode

Garik098, 2021-11-23:
In Pump Configuration Control TC-PCC-2.2 and TC-PCC-2.3 the values we write to operation mode and control mode are not reflecting in the effective operation mode and effective control mode. As per the test case we need to write to operation mode and control mode and the same values should reflect in effective operation mode and effective control mode, but it is not happening. 

But I see the operation mode and control mode is showing up the value but not effective operation or control modes. Below is the log reflecting the same.

adk@adks-Mac-mini debug % ./chip-tool pumpconfigurationandcontrol write operation-mode 1 1 1 
[1637654772031] [28386:2663476] CHIP: [CTL] Generating NOC
[1637654772031] [28386:2663476] CHIP: [CTL] Generating ICAC
[1637654772031] [28386:2663476] CHIP: [DL] AUDIT: ===== RANDOM NUMBER GENERATOR AUDIT START ====
[1637654772031] [28386:2663476] CHIP: [DL] AUDIT: * Validate buf1 and buf2 are <<<different every run/boot!>>>
[1637654772031] [28386:2663476] CHIP: [DL] AUDIT: * Validate r1 and r2 are <<<different every run/boot!>>>
[1637654772031] [28386:2663476] CHIP: [DL] AUDIT: * buf1: 3803136E779AB65BFA8143C5B6C9171E
[1637654772031] [28386:2663476] CHIP: [DL] AUDIT: * buf2: E7507482D60A9CE23C478119FD3BD958
[1637654772031] [28386:2663476] CHIP: [DL] AUDIT: * r1: 0xF9BC8F25 r2: 0x85AFE51D
[1637654772031] [28386:2663476] CHIP: [DL] AUDIT: ===== RANDOM NUMBER GENERATOR AUDIT END ====
[1637654772031] [28386:2663476] CHIP: [DL] _Init
[1637654772031] [28386:2663476] CHIP: [IN] UDP::Init bind&listen port=5541
[1637654772032] [28386:2663476] CHIP: [IN] UDP::Init bound to port=5541
[1637654772032] [28386:2663476] CHIP: [IN] UDP::Init bind&listen port=5541
[1637654772032] [28386:2663476] CHIP: [IN] UDP::Init bound to port=5541
[1637654772032] [28386:2663476] CHIP: [IN] TransportMgr initialized
[1637654772032] [28386:2663476] CHIP: [DIS] Init fabric pairing table with server storage
[1637654772032] [28386:2663476] CHIP: [CTL] System State Initialized...
[1637654772032] [28386:2663476] CHIP: [ZCL] Using ZAP configuration...
[1637654772033] [28386:2663476] CHIP: [DIS] Verifying the received credentials
[1637654772034] [28386:2663476] CHIP: [DIS] Added new fabric at index: 1, Initialized: 1
[1637654772034] [28386:2663476] CHIP: [DIS] Assigned compressed fabric ID: 0x56791D9DD63C9A95, node ID: 0x000000000001B669
[1637654772034] [28386:2663476] CHIP: [CTL] Joined the fabric at index 1. Compressed fabric ID is: 0x56791D9DD63C9A95
[1637654772034] [28386:2663479] CHIP: [DL] _OnPlatformEvent
[1637654772034] [28386:2663479] CHIP: [TOO] Sending command to node 0x1
[1637654772034] [28386:2663479] CHIP: [IN] Prepared plaintext message 0x7feab580a680 to 0x0000000000000000 of type 0x30 and protocolId (0, 0) on exchange 34471i with MessageCounter:1455612861.
[1637654772034] [28386:2663479] CHIP: [IN] Sending plaintext msg 0x7feab580a680 with MessageCounter:1455612861 to 0x0000000000000000 at monotonic time: 469669882 msec
[1637654772034] [28386:2663479] CHIP: [SC] Sent Sigma1 msg
[1637654772052] [28386:2663480] CHIP: [EM] Received message of type 0x31 with protocolId (0, 0) and MessageCounter:2116515986 on exchange 34471i
[1637654772052] [28386:2663480] CHIP: [EM] Rxd Ack; Removing MessageCounter:1455612861 from Retrans Table on exchange 34471i
[1637654772052] [28386:2663480] CHIP: [EM] Removed CHIP MessageCounter:1455612861 from RetransTable on exchange 34471i
[1637654772052] [28386:2663480] CHIP: [SC] Received Sigma2 msg
[1637654772052] [28386:2663480] CHIP: [SC] Peer assigned session session ID 16
[1637654772053] [28386:2663480] CHIP: [SC] Sending Sigma3
[1637654772053] [28386:2663480] CHIP: [EM] Piggybacking Ack for MessageCounter:2116515986 on exchange: 34471i
[1637654772053] [28386:2663480] CHIP: [IN] Prepared plaintext message 0x7feab580a680 to 0x0000000000000000 of type 0x32 and protocolId (0, 0) on exchange 34471i with MessageCounter:1455612862.
[1637654772053] [28386:2663480] CHIP: [IN] Sending plaintext msg 0x7feab580a680 with MessageCounter:1455612862 to 0x0000000000000000 at monotonic time: 469669902 msec
[1637654772054] [28386:2663480] CHIP: [SC] Sent Sigma3 msg
[1637654772064] [28386:2663480] CHIP: [EM] Received message of type 0x40 with protocolId (0, 0) and MessageCounter:2116515987 on exchange 34471i
[1637654772065] [28386:2663480] CHIP: [EM] Rxd Ack; Removing MessageCounter:1455612862 from Retrans Table on exchange 34471i
[1637654772065] [28386:2663480] CHIP: [EM] Removed CHIP MessageCounter:1455612862 from RetransTable on exchange 34471i
[1637654772065] [28386:2663480] CHIP: [SC] Success status report received. Session was established
[1637654772065] [28386:2663480] CHIP: [IN] New secure session created for device 0x0000000000000001, key 16!!
[1637654772065] [28386:2663480] CHIP: [TOO] Sending cluster (0x0200) command (0x01) on endpoint 1
[1637654772066] [28386:2663480] CHIP: [DMG] WriteClient moving to [Initialize]
[1637654772066] [28386:2663480] CHIP: [DMG] WriteClient moving to [AddAttribu]
[1637654772066] [28386:2663480] CHIP: [IN] Prepared encrypted message 0x7feab580a680 to 0x0000000000000001 of type 0x6 and protocolId (0, 1) on exchange 34472i with MessageCounter:0.
[1637654772066] [28386:2663480] CHIP: [IN] Sending encrypted msg 0x7feab580a680 with MessageCounter:0 to 0x0000000000000001 at monotonic time: 469669915 msec
[1637654772067] [28386:2663480] CHIP: [DMG] WriteClient moving to [AwaitingRe]
[1637654772067] [28386:2663480] CHIP: [EM] Sending Standalone Ack for MessageCounter:2116515987 on exchange 34471i
[1637654772067] [28386:2663480] CHIP: [IN] Prepared plaintext message 0x70000f2ae450 to 0x0000000000000000 of type 0x10 and protocolId (0, 0) on exchange 34471i with MessageCounter:1455612863.
[1637654772067] [28386:2663480] CHIP: [IN] Sending plaintext msg 0x70000f2ae450 with MessageCounter:1455612863 to 0x0000000000000000 at monotonic time: 469669915 msec
[1637654772067] [28386:2663480] CHIP: [EM] Flushed pending ack for MessageCounter:2116515987 on exchange 34471i
[1637654772071] [28386:2663480] CHIP: [EM] Received message of type 0x7 with protocolId (0, 1) and MessageCounter:1 on exchange 34472i
[1637654772071] [28386:2663480] CHIP: [EM] Rxd Ack; Removing MessageCounter:0 from Retrans Table on exchange 34472i
[1637654772071] [28386:2663480] CHIP: [EM] Removed CHIP MessageCounter:0 from RetransTable on exchange 34472i
[1637654772071] [28386:2663480] CHIP: [DMG] WriteResponse =
[1637654772071] [28386:2663480] CHIP: [DMG] {
[1637654772071] [28386:2663480] CHIP: [DMG] 	AttributeStatusList =
[1637654772071] [28386:2663480] CHIP: [DMG] 	[
[1637654772071] [28386:2663480] CHIP: [DMG] 		AttributeStatusIB =
[1637654772071] [28386:2663480] CHIP: [DMG] 		{
[1637654772071] [28386:2663480] CHIP: [DMG] 			AttributePath =
[1637654772071] [28386:2663480] CHIP: [DMG] 			{
[1637654772071] [28386:2663480] CHIP: [DMG] 				FieldTag = 0x0000_0020,
[1637654772071] [28386:2663480] CHIP: [DMG] 				NodeId = 0x0,
[... 188 lines cut ...]
[1637654784036] [28388:2663532] CHIP: [DMG] 				
[1637654784036] [28388:2663532] CHIP: [DMG] 			Data = 1, 
[1637654784036] [28388:2663532] CHIP: [DMG] 			DataElementVersion = 0x0,
[1637654784036] [28388:2663532] CHIP: [DMG] 		},
[1637654784036] [28388:2663532] CHIP: [DMG] 		
[1637654784036] [28388:2663532] CHIP: [DMG] 	],
[1637654784036] [28388:2663532] CHIP: [DMG] 	
[1637654784036] [28388:2663532] CHIP: [DMG] }
[1637654784036] [28388:2663532] CHIP: [ZCL] ReadAttributesResponse:
[1637654784036] [28388:2663532] CHIP: [ZCL]   ClusterId: 0x0000_0200
[1637654784036] [28388:2663532] CHIP: [ZCL]   attributeId: 0x0000_0020
[1637654784036] [28388:2663532] CHIP: [ZCL]   status: Success                (0x0000)
[1637654784036] [28388:2663532] CHIP: [ZCL]   attribute TLV Type: 0x04
[1637654784036] [28388:2663532] CHIP: [TOO] Int8u attribute Response: 1
[1637654784036] [28388:2663532] CHIP: [EM] Piggybacking Ack for MessageCounter:1 on exchange: 45101i
[1637654784036] [28388:2663532] CHIP: [IN] Prepared encrypted message 0x7ff5cd80a680 to 0x0000000000000001 of type 0x1 and protocolId (0, 1) on exchange 45101i with MessageCounter:1.
[1637654784036] [28388:2663532] CHIP: [IN] Sending encrypted msg 0x7ff5cd80a680 with MessageCounter:1 to 0x0000000000000001 at monotonic time: 469681884 msec
[1637654784036] [28388:2663532] CHIP: [DMG] Client[0] moving to [UNINIT]
[1637654784036] [28388:2663527] CHIP: [CTL] Shutting down the commissioner
[1637654784036] [28388:2663527] CHIP: [CTL] Shutting down the controller
[1637654784037] [28388:2663527] CHIP: [CTL] Shutting down the System State, this will teardown the CHIP Stack
[1637654784037] [28388:2663527] CHIP: [DL] Inet Layer shutdown
[1637654784037] [28388:2663527] CHIP: [DL] BLE layer shutdown
[1637654784037] [28388:2663527] CHIP: [DL] System Layer shutdown
adk@adks-Mac-mini debug %

Garik098, 2022-01-25:
Still seeing the above issue on TE7.5 testing

vivien-apple, 2022-04-05:
I suspect that the PCC code is not implemented. Looking into https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/pump-configuration-and-control-server/pump-configuration-and-control-server.cpp it just seems to do nothing.

Is the intended behaviour for it is to be implemented on a per-vendor basis ? (That would not be surprising).

closed 2022-04-11 by andy31415