project-chip/connectedhomeip #12174: KeyValueStoreMgr().Get() fails on Darwing

rcasallas-silabs, 2021-11-23:
#### Problem
This behaviour has been observed on the Darwing implementation of KeyValueStoreMgr:
* If more than one new key is store with KeyValueStoreMgr().Put(), the following KeyValueStoreMgr().Get() call for the second key returns CHIP_NO_ERROR,  but fails to retrieve the stored value (the output buffer remains unchanged).
* However, if KeyValueStoreMgr().Get() is called immediatelly after the KeyValueStoreMgr().Put() call, the value is later returned correctly.
* In any case, the KeyValueStoreMgr().Get() call does not modify the value of the in/out size parameter.
The following code illustrate the problem:
```

static constexpr size_t kBufferSize = 512;
static constexpr size_t kEntriesCount = 4;
static constexpr size_t kEntrySizeMax = 128;
static constexpr size_t kKeySizeMax = 16;

void FormatKey(char *key, size_t max_size, size_t index)
{
    snprintf(key, max_size, "e%zu", index);
}

void TestKeyValueStoreMgr(nlTestSuite * apSuite, void * apContext)
{
    static uint8_t buffer[kBufferSize] = { 0 };
    ByteSpan entries[kEntriesCount];
    char key[kKeySizeMax];
    uint8_t out[kEntrySizeMax];
    size_t out_size;

    // Initialize

    for (size_t i = 0; i < sizeof(buffer); i++)
    {
        buffer[i] = static_cast<uint8_t>(i % 0xff);
    }
    
    // Delete

    for (size_t i = 0; i < kEntriesCount; i++)
    {
        FormatKey(key, sizeof(key), i);
        NL_TEST_ASSERT(apSuite, CHIP_NO_ERROR == chip::DeviceLayer::PersistedStorage::KeyValueStoreMgr().Delete(key));
    }

    // Save

    for (size_t i = 0; i < kEntriesCount; i++)
    {
        FormatKey(key, sizeof(key), i);
        entries[i] = ByteSpan(&buffer[16 + i], (1 + (i % kEntrySizeMax)));
        NL_TEST_ASSERT(apSuite, CHIP_NO_ERROR == chip::DeviceLayer::PersistedStorage::KeyValueStoreMgr().Put(key, entries[i].data(), entries[i].size()));
        out_size = sizeof(out);
        // FIXME: Loading will fail if the following line is commented
        // NL_TEST_ASSERT(apSuite, CHIP_NO_ERROR == chip::DeviceLayer::PersistedStorage::KeyValueStoreMgr().Get(key, out, out_size));
    }

    // Load

    for (size_t i = 0; i < kEntriesCount; i++)
    {
        FormatKey(key, sizeof(key), i);
        out_size = sizeof(out);
        memset(out, 0x00, out_size);
        NL_TEST_ASSERT(apSuite, CHIP_NO_ERROR == chip::DeviceLayer::PersistedStorage::KeyValueStoreMgr().Get(key, out, out_size));
        NL_TEST_ASSERT(apSuite, !memcmp(out, entries[i].data(), entries[i].size()));
    }
}
```

 #### Proposed Solution
None so far.

bzbarsky-apple, 2021-12-01:
@sagar-apple Is this code you know about?  Or does @pan-apple ?

bzbarsky-apple, 2022-01-25:
It's possible that https://github.com/project-chip/connectedhomeip/pull/14233 will fix this.  At least it fixes the above-pasted test for me....

closed 2022-01-27 by bzbarsky-apple