project-chip/connectedhomeip #12190: Global attribute " ClusterRevision " default value mismatch

manjunathambiger23, 2021-11-24:
Summary:
                Reading default value of " ClusterRevision " attribute showing mismatch when made a run for the testacases below 
TC-CC-1.1 ,TC-LVL-1.1, TC-OCC-1.1, TC-PCC-1.1, TC-RH-1.1, TC-TM-1.1, TC-TSTAT-1.1, TC-TSUIC-1.1 

Spec Screenshot:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Testplan Screenshot:
[image]

Execution logs:
[CC.txt](https://github.com/project-chip/connectedhomeip/files/7593082/CC.txt)
[LVL.txt](https://github.com/project-chip/connectedhomeip/files/7593083/LVL.txt)
[MC.txt](https://github.com/project-chip/connectedhomeip/files/7593084/MC.txt)
[OCC.txt](https://github.com/project-chip/connectedhomeip/files/7593085/OCC.txt)
[PCC.txt](https://github.com/project-chip/connectedhomeip/files/7593086/PCC.txt)
[RHM.txt](https://github.com/project-chip/connectedhomeip/files/7593087/RHM.txt)
[TM.txt](https://github.com/project-chip/connectedhomeip/files/7593088/TM.txt)
[TSTAT2.txt](https://github.com/project-chip/connectedhomeip/files/7593089/TSTAT2.txt)
[TSUIC.txt](https://github.com/project-chip/connectedhomeip/files/7593090/TSUIC.txt)

sowmyassp, 2022-02-08:
Even in Flow Measurement cluster [TC-FLW-1.1] test case we are facing same issue  

[image]
Execution log:
[TC_FLW_1.1_cntrl.txt](https://github.com/project-chip/connectedhomeip/files/8021958/TC_FLW_1.1_cntrl.txt)

Rajashreekalmane, 2022-02-08:
Even in  Illuminance Measurement cluster [TC-ILL-1.1] test case we are facing same issue.

[image]

PFA Test log
[Uploading [TC-ILL-1.1].txt…]()

somu1710, 2022-02-23:
https://github.com/project-chip/connectedhomeip/issues/14628

Survensa, 2022-03-22:
Same issue is reflected in thread connectivity.

andy31415, 2022-04-14:
I believe we have to check cluster revision defaults in the SDK and then compare with spec. We should currently implement the full spec so values should match exactly. For reference, this can output cluster revision values for zap files using https://stedolan.github.io/jq/:

```
jq '.endpointTypes[].clusters[] | {name: .name, attr: .attributes[] | {name:.name, default:.defaultValue}} | select(.attr.name == "ClusterRevision" )' ./examples/lighting-app/lighting-common/lighting-app.zap
```

andy31415, 2022-04-14:
This is more compact:

```
jq '.endpointTypes[].clusters[] | {name: .name, attr: .attributes[] | {name:.name, default:.defaultValue}} | select(.attr.name == "ClusterRevision") | {name:.name, clusterRevision: .attr.default}' ./examples/lighting-app/lighting-common/lighting-app.zap
```

andy31415, 2022-04-14:
Pretty output:

```
jq '.endpointTypes[].clusters[] | {name: .name, attr: .attributes[] | {name:.name, default:.defaultValue}} | select(.attr.name == "ClusterRevision") | {name:.name, clusterRevision: .attr.default} | [.name + ":" + .clusterRevision] | @tsv' ./examples/lighting-app/lighting-common/lighting-app.zap | sort | uniq | sed 's#"\(.*\)"#\1#' | column -s: -t
``` 

this was painful to figure out, but looks like:

```
Access Control                 1
AdministratorCommissioning     0x0001
Basic                          1
Color Control                  3
Color Control                  5
Descriptor                     1
Ethernet Network Diagnostics   0x0001
Fixed Label                    1
General Commissioning          0x0001
General Diagnostics            0x0001
Group Key Management           1
Groups                         3
Identify                       4
Level Control                  5
Localization Configuration     1
Network Commissioning          0x0001
Occupancy Sensing              2
On/Off                         2
On/Off                         4
On/off Switch Configuration    0x0001
Operational Credentials        0x0001
OTA Software Update Provider   0x0001
OTA Software Update Requestor  1
Scenes                         3
Software Diagnostics           0x0001
Thread Network Diagnostics     0x0001
Time Format Localization       1
Unit Localization              1
User Label                     1
WiFi Network Diagnostics       0x0001
```

Notice how On/Off cluster has 2 separate revisions and so does color control and maybe others. We have to go through all revisions on all clusters and match with spec.

andy31415, 2022-04-14:
All current revisions in all zap files based on (ran in fish shell):

```
for name in (fd .zap | grep -v zapt | grep -v third_party); jq '.endpointTypes[].clusters[] | {name: .name, attr: .attributes[] | {name:.name, default:.defaultValue}} | select(.attr.name == "ClusterRevision") | {name:.name, clusterRevision: .attr.default} | [.name + ":" + .clusterRevision] | @tsv' $name | sort | uniq | sed 's#"\(.*\)"#\1#'; end | sort | uniq | column -s: -t
```

```
Access Control                           1
Account Login                            0x0001
AdministratorCommissioning               0x0001
AdministratorCommissioning               1
Alarms                                   0x0001
Application Basic                        0x0001
Application Basic                        1
Application Launcher                     0x0001
Audio Output                             0x0001
Barrier Control                          0x0001
Barrier Control                          1
Basic                                    1
Binary Input (Basic)                     0x0001
Binding                                  0x0001
Binding                                  1
Boolean State                            1
Bridged Actions                          1
Bridged Device Basic                     1
Channel                                  0x0001
Color Control                            3
Color Control                            5
Content Launcher                         0x0001
Content Launcher                         1
Descriptor                               0x0001
Descriptor                               1
Device Temperature Configuration         0x0001
Diagnostic Logs                          1
Door Lock                                3
Electrical Measurement                   3
Ethernet Network Diagnostics             0x0001
Fan Control                              1
Fixed Label                              0x0001
Fixed Label                              1
Flow Measurement                         0x0001
Flow Measurement                         2
General Commissioning                    0x0001
General Commissioning                    1
General Diagnostics                      0x0001
Group Key Management                     0x0001
Group Key Management                     1
Groups                                   3
IAS Zone                                 2
Identify                                 2
Identify                                 4
Illuminance Measurement                  2
Keypad Input                             0x0001
Keypad Input                             1
Level Control                            5
Localization Configuration               1
Low Power                                0x0001
Media Input                              0x0001
Media Playback                           0x0001
Mode Select                              1
Network Commissioning                    0x0001
Network Commissioning                    1
Occupancy Sensing                        2
On/Off                                   2
On/Off                                   3
On/Off                                   4
On/off Switch Configuration              0x0001
On/off Switch Configuration              1
Operational Credentials                  0x0001
Operational Credentials                  1
OTA Software Update Provider             0x0001
OTA Software Update Provider             1
OTA Software Update Requestor            0x0001
OTA Software Update Requestor            1
Power Configuration                      2
Power Source                             1
Power Source Configuration               1
Pressure Measurement                     2
Pump Configuration and Control           0x0001
Pump Configuration and Control           1
Relative Humidity Measurement            1
Relative Humidity Measurement            2
Scenes                                   3
Software Diagnostics                     0x0001
Switch                                   0x0001
Switch                                   1
Target Navigator                         0x0001
Target Navigator                         1
Temperature Measurement                  0x0001
Temperature Measurement                  1
Temperature Measurement                  3
Test Cluster                             0x0001
Thermostat                               3
Thermostat                               5
Thermostat User Interface Configuration  0x0001
Thermostat User Interface Configuration  1
Thread Network Diagnostics               0x0001
Time Format Localization                 1
Unit Localization                        1
User Label                               1
Wake on LAN                              0x0001
WiFi Network Diagnostics                 0x0001
Window Covering                          3
Window Covering                          5
```

closed 2022-04-15 by Damian-Nordic

somu1710, 2022-04-19:
Relative humidity still throws a old cluster revision value as 2

PR #17410 (2022-04-15): Update revision default values to spec 
#### Problem
Cluster revision values do not match spec.

#### Change overview
Match spec as much I could double-check. Also remove hex encoding for spec numbers (makes no sense).

Fixes #12190

Note that this was not done manually (that would be way too hard) instead a script was used, relevant input looking like:

```py
        replacers = []
        replacers.append(ClusterRevisionReplacer(cluster_name="Access Control", code=31, revision=1))
        replacers.append(ClusterRevisionReplacer(cluster_name="Account Login", code=1294, revision=1))
        r [... cut]