project-chip/connectedhomeip #12196: TvChannelManager Cluster errors

xylophone21, 2021-11-24:
#### Problem
 * read tv-channel-lineup, current-tv-channel  not implemented yet
     * If added in my local with aEncoder.Encode(channelLineupInfo) and the chip-tool will response Wrong TLV type 
 * ChangeChannel Request has no response
     * src/app/clusters/tv-channel-server/tv-channel-server.cpp, // TODO: Enable this once struct as param is supported

xylophone21, 2021-11-24:
logs for read tv-channel-lineup

[1637755782706] [2809:25148] CHIP: [EM] Received message of type 0x5 with protocolId (0, 1) and MessageCounter:1 on exchange 56081i
[1637755782707] [2809:25148] CHIP: [EM] Rxd Ack; Removing MessageCounter:1 from Retrans Table on exchange 56081i
[1637755782707] [2809:25148] CHIP: [EM] Removed CHIP MessageCounter:1 from RetransTable on exchange 56081i
[1637755782707] [2809:25148] CHIP: [DMG] ReportDataMessage =
[1637755782707] [2809:25148] CHIP: [DMG] {
[1637755782707] [2809:25148] CHIP: [DMG]        AttributeReportIBs =
[1637755782707] [2809:25148] CHIP: [DMG]        [
[1637755782707] [2809:25148] CHIP: [DMG]                AttributeReportIB =
[1637755782707] [2809:25148] CHIP: [DMG]                {
[1637755782707] [2809:25148] CHIP: [DMG]                        AttributeDataIB =
[1637755782707] [2809:25148] CHIP: [DMG]                        {
[1637755782707] [2809:25148] CHIP: [DMG]                                AttributePathIB =
[1637755782707] [2809:25148] CHIP: [DMG]                                {
[1637755782707] [2809:25148] CHIP: [DMG]                                        Endpoint = 0x1,
[1637755782707] [2809:25148] CHIP: [DMG]                                        Cluster = 0x504,
[1637755782707] [2809:25148] CHIP: [DMG]                                        Attribute = 0x0000_0001,
[1637755782707] [2809:25148] CHIP: [DMG]                                }
[1637755782707] [2809:25148] CHIP: [DMG] 
[1637755782707] [2809:25148] CHIP: [DMG]                                        Data = 
[1637755782707] [2809:25148] CHIP: [DMG]                                        {
[1637755782707] [2809:25148] CHIP: [DMG]                                                0x0 = "postalCo", 
[1637755782707] [2809:25148] CHIP: [DMG]                                                0x1 = "lineup", 
[1637755782707] [2809:25148] CHIP: [DMG]                                                0x2 = "postalCode", 
[1637755782707] [2809:25148] CHIP: [DMG]                                                0x3 = 0, 
[1637755782707] [2809:25148] CHIP: [DMG]                                        },
[1637755782707] [2809:25148] CHIP: [DMG]                                DataVersion = 0x0,
[1637755782707] [2809:25148] CHIP: [DMG]                        },
[1637755782707] [2809:25148] CHIP: [DMG] 
[1637755782707] [2809:25148] CHIP: [DMG]                },
[1637755782707] [2809:25148] CHIP: [DMG] 
[1637755782707] [2809:25148] CHIP: [DMG]        ],
[1637755782707] [2809:25148] CHIP: [DMG] 
[1637755782707] [2809:25148] CHIP: [DMG] }
[1637755782707] [2809:25148] CHIP: [ZCL] ReadAttributesResponse:
[1637755782707] [2809:25148] CHIP: [ZCL]   ClusterId: 0x0000_0504
[1637755782707] [2809:25148] CHIP: [ZCL]   attributeId: 0x0000_0001
[1637755782707] [2809:25148] CHIP: [ZCL]   status: Success                (0x0000)
[1637755782707] [2809:25148] CHIP: [ZCL]   attribute TLV Type: 0x15
[1637755782707] [2809:25148] CHIP: [ZCL] Failed to get value from TLV data for attribute reading response: ../../examples/chip-tool/third_party/connectedhomeip/src/lib/core/CHIPTLVReader.cpp:398: CHIP Error 0x00000026: Wrong TLV type
[1637755782707] [2809:25148] CHIP: [TOO] Default Failure Response: 0x87
[1637755782711] [2809:25148] CHIP: [EM] Piggybacking Ack for MessageCounter:1 on exchange: 56081i
[1637755782711] [2809:25148] CHIP: [IN] Prepared encrypted message 0x145812680 to 0x000000000001B669 of type 0x1 and protocolId (0, 1) on exchange 56081i with MessageCounter:2.
[1637755782711] [2809:25148] CHIP: [IN] Sending encrypted msg 0x145812680 with MessageCounter:2 to 0x000000000001B669 at monotonic time: 1346794 msec
[1637755782711] [2809:25148] CHIP: [DMG] Client[0] moving to [UNINIT]
[1637755782711] [2809:25143] CHIP: [-] ../../examples/chip-tool/third_party/connectedhomeip/zzz_generated/chip-tool/zap-generated/cluster/Commands.h:2054: CHIP Error 0x000000AC: Internal error at ../../examples/chip-tool/commands/common/CHIPCommand.cpp:84
[1637755782711] [2809:25143] CHIP: [TOO] Run command failure: ../../examples/chip-tool/third_party/connectedhomeip/zzz_generated/chip-tool/zap-generated/cluster/Commands.h:2054: CHIP Error 0x000000AC: Internal error
[1637755782737] [2809:25143] CHIP: [CTL] Shutting down the System State, this will teardown the CHIP Stack
[1637755782737] [2809:25143] CHIP: [DL] Inet Layer shutdown
[1637755782737] [2809:25143] CHIP: [DL] BLE shutdown
[1637755782737] [2809:25143] CHIP: [DL] System Layer shutdown

franck-apple, 2022-01-22:
Please file this issue in the Test Harness repo. Closing it here.

closed 2022-10-18 by stale[bot]