project-chip/connectedhomeip #12216: Commissioner discovery over DNS-SD returns no results

sharadb-amazon, 2021-11-24:
#### Problem
* Expected behavior: tv-casting-app should be able to discover on network commissioners using DNS-SD and list them to the user on startup
* Actual Behavior: tv-casting-app, does NOT discover any commissioners
* Steps to reproduce:
1. Build and run a commissioner that advertises itself over DNS-SD like _chip-tv-app_
2. Build and run the _chip-tv-casting-app_
3. Note that the console output on the tv-casting-app's side has a line stating: "CHIP:SVR: No commissioner discovered, commissioning must be initiated manually!" like the one below:
```
$ ./out/host/chip-tv-casting-app
[1637777886.417526][2375:2375] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-2Kct3K)
[1637777886.420274][2375:2375] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1637777886.420446][2375:2375] CHIP:DL: NVS set: chip-counters/reboot-count = 54 (0x36)
[1637777886.424698][2375:2375] CHIP:DL: Failed to get Ethernet interface
[1637777886.425141][2375:2375] CHIP:DL: Failed to reset Ethernet statistic counts
[1637777886.425390][2375:2375] CHIP:DL: Failed to get WiFi interface
[1637777886.425497][2375:2375] CHIP:DL: Failed to reset WiFi statistic counts
[1637777886.439353][2375:2375] CHIP:DIS: Attempt to mDNS broadcast failed:  ../../third_party/connectedhomeip/src/inet/UDPEndPoint.cpp:1044: OS Error 0x02000065: Network is unreachable
[1637777891.442438][2375:2375] CHIP:SVR: No commissioner discovered, commissioning must be initiated manually!
...
...

```
A similar issue is observed when using the _"chip-tool discover commissioners"_ command.

 #### Proposed Solution
TBD?

**NOTE: This issue started happening after the following commit was made: https://github.com/sharadb-amazon/connectedhomeip/commit/7c47bed46b4b7a4596564c751e302390f5eb17f9#diff-4e8fe74ec1a9ccd5056ea231adc7f05806fbedeff1c77e2d9d34bdb88cca689d**

sharadb-amazon, 2021-11-24:
Also note, in the test above, I ran the chip-tv-app and the chip-tv-casting-app on the same docker instance, in case that is relevant.

andy31415, 2021-11-24:
Maybe fixed in #12161 if the error is apps listening on 5353 and conflicting.

When you see the failure, could you run a `netstat -a -n -p | grep 5353` and see how many things are listening on that port? Generally there should be only one but my guess is you have several.

andy31415, 2021-11-24:
Likely running chip-tv-app and chip-tv-casting-app is the cause (both run on 5353 and conflict). You could use network namespaces to run separate networks for each and then your tests may pass (currently test_suites.sh does this for linux test runs).

cecille, 2021-11-24:
I suspect this is related to https://github.com/project-chip/connectedhomeip/pull/11635. Can you confirm 
- which mdns implementation is being used (platform or minimal) on both the commissioner and the app?
- what os you're using to run these (both on linux, both on mac, separate devices etc.)
- whether you're running another mdns instance alongside these (avahi, bonjour etc.)

I agree with Andrei - the PR he linked may be the answer here. Another option is to force both to use the platform mdns implementation (chip_mdns="platform")

sharadb-amazon, 2021-11-24:
https://github.com/project-chip/connectedhomeip/pull/12161 fixed the issue.

closed 2021-11-24 by sharadb-amazon