project-chip/connectedhomeip #12249: Circular dependencies with lwip

denravonska, 2021-11-25:
#### Problem
I have a project which will use Matter with lwip enabled.

We use the lib as a submodule to our project with the following structure:
```
args.gni
build -> third_party/connectedhomeip/build
BUILD.gn
build_overrides -> third_party/connectedhomeip/examples/build_overrides
config -> third_party/connectedhomeip/config
third_party/
├─ connectedhomeip
```

As a test I added `"//third_party/connectedhomeip/src/system"` as a dependency to our project and I get this on `master@c79b67bc73ea4bdd2fb70c461e4ca1fc944aac8d`:

```
$ gn gen out
ERROR Dependency cycle:
  //third_party/connectedhomeip/src/lib/support:support ->
  //third_party/connectedhomeip/src/lib/core:chip_config_header ->
  //third_party/connectedhomeip/src/system:system_config_header ->
  //third_party/connectedhomeip/src/lwip:lwip ->
  //third_party/connectedhomeip/src/lib/support:support
```

Is there anything we should configure in our project?

holbrookt, 2022-01-31:
@denravonska I'm helping triage bugs that are critical for 1.0. Is this still an issue for you?

denravonska, 2022-02-01:
@holbrookt Just tested this with `da22674736ebd5b5d88463b0a84896dbad70c20a` and it seems like I can't reproduce it anymore. I can reopen this if it pops up again further down the line.