project-chip/connectedhomeip #12258: [ble connect]controller commissioning dut fail by ble

Suguotao, 2021-11-25:
dut:k32w06
controller:raspbeerypi 4B+ 
#### Problem
 * expected behavior
    commission successfully
 * actual behavior
     commission faily
 * steps to reproduce
   a. cd  /home/ubuntu/connectedhomeip
   b. source ./out/python_env/bin/activate
   c. chip-device-ctrl
   d. connect -ble 3840 20202021 12344321

   then ,i found dut will crash，can't work
 * screenshots, images, logs, etc.
 >
[controller_log.txt](https://github.com/project-chip/connectedhomeip/files/7601779/controller_log.txt)
[dut_log.txt](https://github.com/project-chip/connectedhomeip/files/7601780/dut_log.txt)

fuxiaoming-lumi, 2021-11-26:
Let me add more details.

fuxiaoming-lumi, 2021-11-26:
DUT: k32w06 DK (Branch TE7 patch this commit )
:raspbeerypi 4B+

Problem
expected behavior
commission successfully

actual behavior
commission faily

steps to reproduce
a. cd /home/ubuntu/connectedhomeip
b. source ./out/python_env/bin/activate
c. chip-device-ctrl
d. connect -ble 3840 20202021 12344321

then ,i found dut will crash，can't work

screenshots, images, logs, etc.

controller_log.txt
dut_log.txt

cecille, 2022-02-01:
Is this still happening on master?

woody-apple, 2022-02-08:
Please re-verify and re-open if this is still happening!

closed 2022-02-08 by woody-apple

fuxiaoming-lumi, 2022-02-08:
After many tests, it is confirmed that it is caused by a hardware problem. We can close this ticket.