project-chip/connectedhomeip #12326: Fortify error on Android Pie

milezhacks, 2021-11-29:
#### Problem
When running CHIP on Android Pie, I'm hitting the following error:
```
FORTIFY: sendto: prevented 40-byte read from 24-byte buffer
```
I've traced this to the call `getifaddrs(&mAddrsList)` on https://github.com/project-chip/connectedhomeip/blob/master/src/inet/InetInterface.cpp#L653

The implementation for this function (on Android) comes from: 
https://github.com/project-chip/connectedhomeip/blob/master/src/inet/InetInterface.cpp#L54

I did NOT encounter this error on Android N.  Incidentally, the Fortify check for send()/sendto() was added in Android O:  https://android.googlesource.com/platform/bionic/+/master/docs/status.md

Upon reviewing the implementation for getifaddrs() in `third_party/android/platform-libcore/android-platform-libcore/ifaddrs-android.h` line 182, it looks like the size of the buffer is swollen by the padding and alignment macros.

Incidentally, bionic has provided its own getifaddrs() since at least Android L.

 #### Proposed Solution
Either:
1. add a max-version check to InetInterface.cpp to NOT use libcore after a certain point
2. OR just use <ifaddrs.h> unconditionally

milezhacks, 2021-11-29:
Here's the diff:
https://github.com/project-chip/connectedhomeip/compare/master...milezhacks:issues/12326?expand=1

Will submit PR once CI completes.

milezhacks, 2021-12-08:
> Incidentally, bionic has provided its own getifaddrs() since at least Android L

This statement is wrong.  getifaddrs() was added to Bionic in Android Nougat.  For Android L (API Level 21) this stub is, in fact, required but shouldn't be used for API levels >= 24.

andy31415, 2022-01-31:
Added looks critical: FORTIFY error points to invalid memory access.