project-chip/connectedhomeip #12328: Add precheck function, isSupportedAttribute, in AttributeAccessInterface

yunhanw-google, 2021-11-29:
currently the code goes to the read for different cluster server, then inside read we start to know if cluster id exists or not
we could introduce isSupportedAttribute in AttributeAccessInterface, it would help avoid subtly different error handling in different server.

turon, 2022-02-01:
Looked at this for v1 triage, and the issue as currently described isn't actionable.
The code referenced in the PR comment doesn't exist anymore or has been severely refactored.
Marking stale.
Removing tag `v`

[src/app/clusters/power-source-configuration-server/power-source-configuration-server.cpp]
```
        }
    }

    return CHIP_NO_ERROR;
 
Member
@tcarmelveilleux tcarmelveilleux on Nov 26, 2021 • 
If we reach here, there was a read path that was not feasible. Can you return an error?

 
Member
Author
@fessehaeve fessehaeve on Nov 27, 2021
Done

 
Member
@tcarmelveilleux tcarmelveilleux on Nov 29, 2021
This always return NO_ERROR, but is it possible there is any fallthrough down to here? How could it happen? I suggest we return an error here rather than CHIP_NO_ERROR.

 
Member
@yunhanw-google yunhanw-google on Nov 29, 2021
agree, better to introduce new error CHIP_ERROR_CLUSTER_NOT_FOUND or something similar
```

bzbarsky-apple, 2022-02-01:
My general take is that it's not clear what the concrete proposal here is.  Right now we guarantee that we only call AttributeAccessInterface for attributes that exist (via the metadata).  Whether they are then _handled_ by the AttibuteAccessInterface is up to it.

closed 2023-03-23 by stale[bot]