project-chip/connectedhomeip #12372: ColorTemperature attribute name is not aligned with spec

joesena-amz, 2021-11-30:
#### Problem
The color temperature value name is misaligned between chip-tool/chip-device-ctrl and the specification.

Spec: ColorTemperatureMireds
[image]

Chip-tool:
`Usage:
  ./chip-tool colorcontrol read attribute-name [param1 param2 ...]

  +-------------------------------------------------------------------------------------+
  | Attributes:                                                                         |
  +-------------------------------------------------------------------------------------+
  | * current-hue                                                                       |
  | * current-saturation                                                                |
  | * remaining-time                                                                    |
  | * current-x                                                                         |
  | * current-y                                                                         |
  | * drift-compensation                                                                |
  | * compensation-text                                                                 |
  | * color-temperature                                                                 |
  | * color-mode                                                                        |
  | * color-control-options                                                             |
  | * number-of-primaries                                                               |
  | * primary1x                                                                         |
  | * primary1y                                                                         |
  | * primary1intensity                                                                 |
  | * primary2x                                                                         |
  | * primary2y                                                                         |
  | * primary2intensity                                                                 |
  | * primary3x                                                                         |
  | * primary3y                                                                         |
  | * primary3intensity                                                                 |
  | * primary4x                                                                         |
  | * primary4y                                                                         |
  | * primary4intensity                                                                 |
  | * primary5x                                                                         |
  | * primary5y                                                                         |
  | * primary5intensity                                                                 |
  | * primary6x                                                                         |
  | * primary6y                                                                         |
  | * primary6intensity                                                                 |
  | * white-point-x                                                                     |
  | * white-point-y                                                                     |
  | * color-point-rx                                                                    |
  | * color-point-ry                                                                    |
  | * color-point-rintensity                                                            |
  | * color-point-gx                                                                    |
  | * color-point-gy                                                                    |
  | * color-point-gintensity                                                            |
  | * color-point-bx                                                                    |
  | * color-point-by                                                                    |
  | * color-point-bintensity                                                            |
  | * enhanced-current-hue                                                              |
  | * enhanced-color-mode                                                               |
  | * color-loop-active                                                                 |
  | * color-loop-direction                                                              |
  | * color-loop-time                                                                   |
  | * color-loop-start-enhanced-hue                                                     |
  | * color-loop-stored-enhanced-hue                                                    |
  | * color-capabilities                                                                |
  | * color-temp-physical-min                                                           |
  | * color-temp-physical-max                                                           |
  | * couple-color-temp-to-level-min-mireds                                             |
  | * start-up-color-temperature-mireds                                                 |
  | * cluster-revision                                                                  |
  +-------------------------------------------------------------------------------------+`

 #### Proposed Solution
Change the spec or align the attribute name with the spec in chip-tool.

bzbarsky-apple, 2022-05-05:
Just to be clear, the issue is that the cluster XML does not match the spec...

bzbarsky-apple, 2022-05-19:
@jmartinez-silabs Could you take a look at aligning the XML with the spec here?  There are all sorts of differences (e.g. the spec has "CurrentY" and the XML has "current y"...

PR #18700 (2022-05-24): Re-align colorcontrol cluster definition with the spec
#### Problem
fixes #12372

#### Change overview
commit 1
- Modify the xmls to align the cluster definitions with the spec
- Apply renames in to yaml test files and color-control server implementation
- relauch zap files so the naming changes are applied
Commit 2
- regen all

#### Testing
Build a few examples  and chip-tool locally.
CI can test against regression through Test_TC_CC_X_X and confirm build for all apps.
closed 2022-05-24 by jmartinez-silabs