project-chip/connectedhomeip #12418: Populate the spec-defined device types supported on a given endpoint for Descriptor.Attributes.DeviceList

yunhanw-google, 2021-12-01:
#### Problem
Currently  spec-defined device types supported has not yet correctly populated on a given endpoind, need to fix it.

mrjerryjohns, 2021-12-06:
With #12612, support for 1 device type per endpoint is complete. Multiple is open however...

woody-apple, 2022-02-08:
@mrjerryjohns Is multiple required for the spec?

leorozendaal, 2022-02-21:
> @mrjerryjohns Is multiple required for the v1 spec?

@woody-apple @mrjerryjohns 
Each endpoint needs (in its Descriptor cluster) a LIST of devicetypestruct (= devicetype+revision).
This is needed for e.g. these cases
- multiple device types on one endpoint (basic composability)
- functional device type on EP0 (next to the Root Node device type)
- bridged devices need an application device type (e.g. Extended Color Light) in addition to the Bridged Node device type on each endpoint.

[image]

bzbarsky-apple, 2022-03-28:
Things we need to do to implement multiple device types for an endpoint:

1. Change the format of the data tables in endpoint_config.h to allow storing more than one device type/revision pair per endpoint.  Probably involves updating `emberAfDeviceIdFromIndex` and `emberAfDeviceVersionFromIndex` to just return the first one. (required)
2. Change `DescriptorAttrAccess::ReadDeviceAttribute` to consume the new data and stop using the current `emberAfDeviceIdFromIndex` and `emberAfDeviceVersionFromIndex`. (required)
3. Change `examples/common/pigweed/rpc_services/Descriptor.h` (optional)
4. Remove `emberAfDeviceIdFromIndex` and `emberAfDeviceVersionFromIndex` (optional)
5. Adding ZAP database support for configuring multiple device types on an endpoint (required, tracked in https://github.com/project-chip/zap/issues/366, can happen in parallel with items 1 and 2).
6. Adding ZAP UI support for configuring multiple device types on an endpoint (required, tracked in https://github.com/project-chip/zap/issues/366, can happen in parallel with items 1 and 2).

mrjerryjohns, 2022-03-29:
Taking this on.

closed 2022-04-07 by mrjerryjohns

mlepage-google, 2022-04-07:
Actually there's one more thing to do, and that is handle device type to endpoint resolving in Access Control. I did this a week or two ago when there was only 1 device type per endpoint, so I'll have to revisit that implementation. I can do so under the auspices of issue #14431.