project-chip/connectedhomeip #12471: OTA is so slow because we call the PollTimerHandler every 500ms

wqx6, 2021-12-02:
#### Problem
The OTA process is so slow. It takes about 5 minutes to transfer a image which is about 150K Bytes. 
The main reason is that we have to wait the `PollTimerHandler` to send the query message or block message even if we have prepared the message. And the `PollTimerHandler` function was called every 500ms by default. That means we will waste a lot of time in BDX transfer.

 #### Proposed Solution
Decrease the Poll Frequency or use a more efficient process of BDX Transfer.

holbrookt, 2022-04-27:
@carol-apple can you take this on or assign it to the right person?

carol-apple, 2022-05-26:
https://github.com/project-chip/connectedhomeip/pull/18377 fixes this

closed 2022-05-26 by carol-apple