project-chip/connectedhomeip #12520: Remove BDX timeout tracking from TransferSession

holbrookt, 2021-12-02:
#### Problem
The `TransferSession` class currently forces the caller to provide timestamps in `PollOutput` and `HandleMessageReceived` in order to determine if an application-determined timeout has occurred. However, the BDX spec does not say that this is required:
```
For synchronous transfers, if the Driver fails to receive the response to any given request that is not received within a particular application-determined time, it SHOULD abort the session.
```
Tracking timeouts in the `TransferSession` class is unnecessary and makes the API more complicated and cumbersome.

 #### Proposed Solution
* put the responsibility on the application to determine if a timeout has occurred
* add a public method `OnTimeout` for the application to call to abort the session (without sending a `StatusReport` like in `Abort()`)
* remove timestamp arguments from aforementioned methods

carol-apple, 2022-03-31:
This actually helps the provider catch the 5 minute idle timeout. If this is removed, there needs to be explicit implementation on the provider side to do the same idle check.

isiu-apple, 2022-04-15:
This issue is being reverted via PR https://github.com/project-chip/connectedhomeip/pull/17427, so re-opening this issue.

nivi-apple, 2023-04-06:
work in progress

cnanoleaf, 2023-10-12:
Hello, any chance for some sort of fix/final solution here? 

We are currently trying to use the provided chip-ota-provider-app to validate a new device side implementation and we are running into a scenario where after the header is sent and received properly, the process just stops as it seems to be waiting for some time to pass on either side.

The code in https://github.com/project-chip/connectedhomeip/pull/16743 seemed promising but then it was reverted?

This code has been officially released as part of the v1.1.0.1 tag.