project-chip/connectedhomeip #12539: Failing Device Attestation Information verification test in ESP32 and NRF Connect platforms

vijs, 2021-12-03:
#### Problem
* Device Attestation Information verification test is failing in ESP32 and NRF Connect platforms
* Reference: [nRF Connect SDK Test #26050](https://github.com/project-chip/connectedhomeip/runs/4399773375?check_suite_focus=true)
 
#### Proposed Solution
* Check if the current time is properly set in the test platform
* Check if the mbedTLS API is working as expected

dhrishi, 2022-01-18:
@vijs Sorry for the delay in getting back. Can you please point to the test case for ESP32, if the issue is still valid?

holbrookt, 2022-01-31:
@vijs please see comment from @dhrishi . We need to know if this is an issue for 1.0 or not.

woody-apple, 2022-02-08:
@vijs Please re-open if this is still happening with more information. Thanks!

closed 2022-02-08 by woody-apple