project-chip/connectedhomeip #12607: Automate the single-Linux-node OTA scenario and add it to CI

selissia, 2021-12-05:
#### Problem
Currently the Requestor-Provider scenario is run as a series of manual commands. We need an ability to run it in a script. In addition, this script must be added to the of GitHub continuous integration for Matter, this will make it a regression test

 #### Proposed Solution

Commands for building the the OTA Requestor, Provider and chip-tool on Linux:
```scripts/examples/gn_build_example.sh examples/chip-tool out/
scripts/examples/gn_build_example.sh examples/ota-provider-app/linux out/debug chip_config_network_layer_ble=false
scripts/examples/gn_build_example.sh examples/ota-requestor-app/linux out/debug```

Steps for running the test scenario:
```rm -r /tmp/chip_*
./out/debug/chip-ota-provider-app -f /tmp/ota.txt
./out/chip-tool pairing onnetwork 1 20202021
./out/debug/chip-ota-requestor-app -u 5560 -d 42 -i ::1
./out/chip-tool pairing onnetwork-long 2 20202021 42
./out/chip-tool otasoftwareupdaterequestor announce-ota-provider 1 0 0 2 0```

Write a script that would run the above build commands and the test scenario. Add the script to `./.github/workflows/examples-linux-standalone.yaml` to make it a part of CI.

carol-apple, 2022-02-07:
Tip: You can mention something like `Fixes: https://github.com/project-chip/connectedhomeip/issues/12607` in the description of this PR and it will close the issue for you once this PR merges.

carol-apple, 2022-02-22:
Fixed by #14414

closed 2022-02-22 by carol-apple