project-chip/connectedhomeip #12627: ReadInteraction.h's ReadAttribute is exploitable as actually used

bzbarsky-apple, 2021-12-06:
#### Problem
ReadAttribute takes two callbacks: a data callback and a failure callback.

It will then proceed to call the data callback for each path in the response.  The consumer has no way to know when the response is done.

Given the way the API is structured, with taking a single concrete path, etc, the only thing the consumer can do is assume it will get a single data callback and clean up after it gets one. But since the actual implementation might deliver multiple callbacks, after the first one we will be calling methods (possibly virtual) on freed objects, which is a recipe for security bugs.

Right now TypedReadAttributeCallback does enforce that the path it gets matches the (single) path being read.  But this still fails in the following cases:

1. A malicious server could send the same path twice, both times with data, causing two calls to onSuccessCb.
2. A malicious server could send the "right" path followed by the "wrong" path.  The first would call onSuccessCb.  The second would fail the "is the right path" check and end up calling onErrorCb.
3. A malicious server could send the same path twice, both times with a failure status, causing two calls to onErrorCb.

 #### Proposed Solution
ReadAttribute needs to either provide a "safe to clean up" callback or enforce that at most one call to at most one of the callbacks it's handed happens.  The latter would be better from the perspective of consumer sanity.

I have not audited carefully, but I am quite sure that InvokeCommandRequest in InvokeInteraction.h and WriteAttribute in WriteInteraction.h have the same problem.  WriteInteraction has provisions for a "done" callback, but allows null and many consumers pass null.

bzbarsky-apple, 2021-12-06:
@mrjerryjohns @yunhanw-google @vivien-apple

erjiaqing, 2021-12-07:
So the expected behavior of `TypedReadAttributeCallback` is 
- Only accept exacttly one AttributeReportIB, and its path should match the path in our record, following calls to OnData / OnError becomes no-op?

bzbarsky-apple, 2021-12-07:
Yes, that would make the most sense to me.  Similar for the invoke and write callbacks.

I can probably fix this myself, fwiw, just not likely to before Friday.

closed 2022-07-02 by woody-apple